eclipse-embed-cdt/eclipse-plugins #185: eclipse Adding cortex-m33 instead of cortex-m3

chaitusvk, 2017-03-05:
i created a project with LPC1768 led blinking eclipse added "-m33" in .cproject in 29t and 105th line

29 th line ----- <option id="ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.family.841291887" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.family" value="ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.mcpu.cortex-m33" valueType="enumerated"/>

105th ---							<option id="ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.family.170359597" superClass="ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.family" value="ilg.gnuarmeclipse.managedbuild.cross.option.arm.target.mcpu.cortex-m33
[cproject.txt](https://github.com/gnuarmeclipse/plug-ins/files/819293/cproject.txt)
[Uploading cproject.txt…]()

" valueType="enumerated"/>

ilg-ul, 2017-03-05:
how did you create the project?

I also created a 'Hello World ARM Cortex-M C/C++ Project' wizard and `-m3` was used, as expected:

```
Building target: m3.elf
Invoking: Cross ARM GNU C++ Linker
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -fno-move-loop-invariants -Wall -Wextra  -g3 -T mem.ld -T libs.ld -T sections.ld -nostartfiles -Xlinker --gc-sections -L"../ldscripts" -Wl,-Map,"m3.map" --specs=nano.specs -o "m3.elf"  ./system/src/newlib/_cxx.o ./system/src/newlib/_exit.o ./system/src/newlib/_sbrk.o ./system/src/newlib/_startup.o ./system/src/newlib/_syscalls.o ./system/src/newlib/assert.o  ./system/src/diag/Trace.o ./system/src/diag/trace_impl.o  ./system/src/cortexm/_initialize_hardware.o ./system/src/cortexm/_reset_hardware.o ./system/src/cortexm/exception_handlers.o  ./system/src/cmsis/system_DEVICE.o ./system/src/cmsis/vectors_DEVICE.o  ./src/Timer.o ./src/_write.o ./src/main.o   
Finished building target: m3.elf```

chaitusvk, 2017-03-14:
sorry for late replay i imported the project of LPC1768 from keil uvision4
and i am adding all header files
i followed steps in eclipse for arm devolepment
<https://www.exploreembedded.com/wiki/index.php?title=Setting%20Up%20ARM%20GCC%20And%20Eclipse%20For%20ExploreM3%20LPC1768>

thank you very much ..

ilg-ul, 2017-03-14:
the link you mention does not explain how to create GNU ARM Eclipse projects. most probably you messed something.

and please do not reply by email, use the web interface (or, at least, do not quote the original message).

TM1234, 2017-03-24:
Hi Liviu 

I created a new project as follows...
File > New > C Project > and then follow the wizard
and when it created it Properties > C/C++ Build > Settings > Tool Settings > Target Processor > ARM family is set to cortex-m33 by default.
Is there any way to override this - e.g. select cortex-m3 from plugin_customization.ini or similar?
I tried exporting the preferences before and after but there was no difference so I can't see an obvious way to capture this setting and enforce it via plugin_customization.ini.

I'm using the 3.3.1 plugins on Eclipse Luna SR2 (32 bit) on Windows 7 64 bit.
Previous versions of the plugins defaulted the project ARM family to cortex-m3 as far as I know.
If you need any more info let me know.

Thanks
Tommy

ilg-ul, 2017-03-24:
ah, now I understand the problem.

I confirm the bug, it was introduced by a careless copy/paste when the list of supported mcus was extended.

I fixed it, please check the beta version 3.4.1-201703241311 from updates-test.

TM1234, 2017-03-24:
Thanks for the quick update Liviu.
I presume that there's no way to override it from the likes of plugin_defaults.ini with 3.3.1?

ilg-ul, 2017-03-24:
I guess not. there were multiple entries marked as defaults, and the last one was m33, so all previous ones were ignored.

TM1234, 2017-03-24:
OK - thanks again.
The way that Eclipse/CDT/plugins store settings whereby some are customizable via plugin_customization.ini etc. but others cannot is confusing and I've never really understood it fully.
I presume that it depends on what API(s) plugins use to retrieve settings or something...?
Anyway thanks for the update/fix and I'll try to pull the 3.4.1 beta and use that.

ilg-ul, 2017-04-05:
please try the version from updates-test

TM1234, 2017-04-05:
> please try the version from updates-test

Sorry - can you elaborate please?
Is updates-test a github or eclispe update site or something else?
I can't see any mention of it on the main GNU ARM Eclipse site.
Apologies if I have overlooked something...

ilg-ul, 2017-04-05:
http://gnuarmeclipse.sourceforge.net/updates-test

TM1234, 2017-04-05:
Thanks - I'll try that tomorrow and report back.

TM1234, 2017-04-05:
I just tried the test version and get the same behaviour.
It looks like when I create a new workspace the Window > Preferences > C/C++ > Debug > GDB option/pane is missing.
It only appears when I create/import a project, compile it and attempt to launch a debug session.
Once I have tried to launch a debug session the option appears.

TM1234, 2017-04-05:
With the updates-test version the default processor is again Cortex-M3 and not Cortex-M33.
Thanks.

ilg-ul, 2017-04-05:
ok, great, we'll close this issue when the official release is out.

closed 2017-04-26 by ilg-ul