eclipse-embed-cdt/eclipse-plugins #187: pyOCD: GDB console doesn't appear

flit, 2017-03-13:
### Description

GDB console doesn't appear when connecting via pyOCD plugin, in either of the Console or Debugger Console views. This is with Neon.2. (Haven't tried other plugins for comparison as I don't have a setup readily available.)

I checked the commit logs back to May 2016 and didn't notice anything obvious that was changed in other plugins but not pyOCD regarding the gdb console. I did see commit 2494e5b7f8 regarding `IGdbDebugConstants.GDB_PROCESS_CREATION_VALUE`, but that didn't seem to be directly related.

Of course, I'm happy to work on a fix, just not sure where to look.

### Steps to Reproduce

1. Create a valid pyOCD launch config.
2. Launch the config.

**Expected behavior:** GDB console appears under Debugger Console.

**Actual behavior:** No GDB console visible.

### Versions

* Plugins 3.4.1, pyOCD 1.1.3
* Neon.2 (4.6.2)
* Java 8 update 65
* macOS 10.11.6
* GNU ARM Embedded gcc 5.4.1 20160919, gdb 7.10.1.20160923

ilg-ul, 2017-03-13:
welcome to Neon.2 'goodies'! ;-)

the GDB console **is** present, but it is now a separate tab. 

the GDB trace console is no longer visible by default, you need to enable it somewhere deep in the preferences pages.

both are considered 'features', according to the CDT team...

anyway, Neon.2 and GDB 7.12 is a no-go.

please retest with Neon.3, to be released soon.

flit, 2017-03-13:
I've tried opening the Debugger Console view, but to no avail. See attached screenshot (this is with an open pyOCD debug session).

(And I can open the GDB trace console by enabling that option, no problem there.)

Anyway, will retest with Neon.3 once it's ready.

[image]

ilg-ul, 2017-03-13:
> I've tried opening the Debugger Console view, but to no avail

you're lucky, at least you saw the console; in my tests, with a freshly installed Eclipse, it does not even show, I had to start the debug sessions for multiple times to see it.

flit, 2017-03-13:
Well, it took some digging for me to find it! It definitely didn't appear automatically.

ilg-ul, 2017-03-13:
If you can find a repetitive sequence to prove it, please open a CDT bug report, otherwise it might never be fixed.

flit, 2017-03-13:
Good idea. I'll do that once Neon.3 is out since it's due shortly. I'll also try to set things up so I can test with the JLink plugin. Do you see the gdb console with JLink?

ilg-ul, 2017-03-13:
> Do you see the gdb console with JLink?

yes, J-Link, QEMU (and, I guess, OpenOCD) are ok.

flit, 2017-03-13:
Well, that's my concern. That there is something missing/different in the pyOCD plugin that causes this problem. I'll try to compare the gdb startup related code between the plugins.

flit, 2017-03-15:
I set up my workspace for CDT development (still Neon.2), so I could try to debug this issue. Running with just the org.eclipse.cdt.gdbjtag.* plugins built locally, others from the CDT install. For whatever reason, the problem is gone now… Closing this issue.

closed 2017-03-15 by flit

ilg-ul, 2017-03-15:
yeah, I'm not surprised by this mysterious behaviour... 

I'm not very happy with the latest CDT development, with each new version I faced more and more problems...

AntonKrug, 2019-01-10:
Good to know, thanks for the feedback.