eclipse-embed-cdt/eclipse-plugins #190: I switched to the latest Version (3.3.1) of the plug-in - now every project is corrupted.

mgiaco, 2017-03-27:
The problem is that when I would like to compile I get the error no make is found. And when i look into the project properties on C/C++ Build I get:

```
Orphaned configuration. No base extension cfg exists for ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug.359419031
``` 

Is there a way to fix that maybe by hand in the *.cproject file?

many thanks
mathias

mgiaco, 2017-03-27:
Okay, now I found out that the update changed my project files *.cproject and *.project. No I tried to revert them and use a fresh import no everthing seems to be correct. I do not know why the plug-in update corrupted these files anyway no it works.

closed 2017-03-27 by mgiaco

ilg-ul, 2017-03-27:
I have serious doubts that the plug-ins update changed anything in .cproject and .project, since the plug-ins themselves have nothing to do directly with these files.

if you have the files on a backup disk, it would be interesting to know what were the changes, perhaps we can identify the culprit.

mgiaco, 2017-03-27:
okay sorry i have already deleted it but if i found the same problem on any project i have I will report it here okay.

ilg-ul, 2017-03-27:
I suggest you install a fresh Eclipse, create a new workspace, and import your existing projects.

generally a broken Eclipse is not worth repairing.