eclipse-embed-cdt/eclipse-plugins #191: Problem associate device to project.

goranlantz, 2017-03-28:
Hi,
When I try to associate a device to a project I get the error message "The currently displayed page contains invalid values". The next time I try to open the "Settings" tab again it will not show up.
However, when I start debugging the device (STM32L432KC) shows up correctly in the L-Link debug plugin configuration and the peripheral view is displayed when starting debugging. I have tried this on NEON and MARS clean installs with GNU ARM Eclipse plugin, GNU ARM toolchain and GNU ARM build tools installed with the same result.
Please help me.
/Göran

ilg-ul, 2017-03-28:
could you provide more details on the exact steps you performed, packages used, versions, etc so I can try to reproduce the problem?

goranlantz, 2017-03-28:
Im am using:
MARS.2
GNU ARM Eclipse 3.3.1. All items in the packet installed.
Downloaded packets: ARM CMSIS, STM32F4XX, STM32F2XX and STM32L4XX. 

when I open the <Project>-Propertiies-C/C++ Build-Settings-Device and select any device I get the error message.

ilg-ul, 2017-03-28:
with the available information I cannot reproduce the problem:

[image]

I tried for both managed and unmanaged projects, and associating devices was ok.

how did you create the project? is it managed or with make? what versions of the packs are you using?

could you temporarily install a separate fresh Eclipse, create a new workspace in the same folder where your original workspace was (to reuse the downloaded packs), create a new project with the wizard, and try to associate the device?

goranlantz, 2017-03-28:
Here is the error message when I try to open "Settings" a second time.

[Pic1.docx](https://github.com/gnuarmeclipse/plug-ins/files/875339/Pic1.docx)

Will try a fresh Eclipse install.

ilg-ul, 2017-03-28:
in my environment I was able to enter the settings page for about 5 times, coming for all possible scenarios, with different landing pages (pages selected when the window was previously closed).

I would say you have a broken setup.

(and docx is a weird choice for passing an image)

closed 2017-04-24 by ilg-ul