eclipse-embed-cdt/eclipse-plugins #192: Includes vanish when switching to AArch64

jmaxx, 2017-03-29:
Once I create a new project, I can edit C/C++ Build -> Settings and select my target platform and everything is fine, however, once I select AArch64 in the Toolchains tab, the includes provided by the eclipse plugin vanish. Of course, I can add those manually, however, I would like to know whether this is a bug or a feature.

My toolchain is gcc-arm-none-eabi-6-2017-q1-update which should support 64 bit (I looked at the code in arm_neon.h for intrinsics and it has the 64bit simd stuff in it.

Thanks a lot in advance!

ilg-ul, 2017-03-29:
> the includes provided by the eclipse plugin vanish

wow! there may be some kind of magic behind the plug-ins! ;-)

I did the following
- created a STM32F407 blinky project
- created a new build configuration named Debug-64, copied from the original Debug
- in the Toolchains tab, I changed the architecture to ARM64
- closed the Properties page
- entered again the Properties page and checked C/C++ General -> Paths and Symbols
- all Includes were exactly where the wizard entered them in the beginning

So, the magic did not work for me.

Please provide detailed steps on how to reproduce the problem.

jmaxx, 2017-03-29:
Thank you for the quick reply! It may very well be the case that the magic is sitting in front of the computer...

When I do the same thing as you, I get the same results.

I started out with a Hello World ARM C++ Project. Initially, that one has several includes. These, however are not listed under C/C++ General -> Paths and Symbols, however I can reference them.
[image]
[image]

That means that my choice of project template is somehow unsuitable I guess!?

ilg-ul, 2017-03-29:
ah, ok, you got confused by the system includes. these are automatically detected by CDT you do not need to explicitly enter them.

but for this you need to have a proper compiler, if CDT cannot start the compiler, you have no system includes.

the project created by the Hello World wizard has no local include folders itself, so it is normal for the window you are mention to be empty, regardless the compiler you are using.

it looks like there is not much magic left...

jmaxx, 2017-03-29:
Got it, thanks a lot for your help and sorry for bothering you!

closed 2017-03-29 by jmaxx

ilg-ul, 2017-03-29:
ok, good luck with your projects!