eclipse-embed-cdt/eclipse-plugins #193: Windows > Preferences > C/C++ > Debug > GDB sometimes doesn't show

TM1234, 2017-04-05:
I'm using the 3.2.1 plugins in Eclipse Luna SR2 32 bit on Window 7 64 bit.
Sometimes (I haven't managed to pin down exactly when yet) the Window > Preferences > C/C++ > Debug > GDB preferences configuration page doesn't show at all - i.e. it doesn't even appear in the Preferences treeview.
Has anybody else noticed this?
I will try to identify when this happens.
It happened just now but after compiling a test program and attempting to launch a debug session the GDB preferences configuration page reappeared.
Thanks a lot.

P.S. I'm using 3.2.1 rather than 3.3,1 because I can't use 3.3.1 due to this issue: https://github.com/gnuarmeclipse/plug-ins/issues/185

TM1234, 2017-04-05:
Related material in this ticket:
https://github.com/gnuarmeclipse/plug-ins/issues/185#issuecomment-291986533

TM1234, 2017-04-05:
I tried the http://gnuarmeclipse.sourceforge.net/updates-test version and get the same behaviour.
When I create a new workspace the Window > Preferences > C/C++ > Debug > GDB preferences option/page is missing.
It only appears after I have created/imported a project, compiled it and have launched a debug session.
After the debug session has been launched (or even attempted and failed - e.g. if a JTAG probe not attached) when I go back to the Preferences dialog I see the GDB option in the treeview and the page appears as expected.

ilg-ul, 2017-04-24:
could you find a way to reproduce the problem with Neon.3?

TM1234, 2017-04-24:
Yes - I will try this when I get a chance - hopefully later this week.

TM1234, 2017-05-01:
Same behaviour with Neon.3 (Windows 32 bit) and GNU ARM Eclipse 3.3.1 on Windos 7 64 bit.
To summarise ... the Windows > Preferences > C/C++ > Debug > GDB  does not show until a debug session run has been attempted after which it does appear.

TM1234, 2018-01-09:
FWIW I still get the same behaviour now on a Windows 64 bit build of the tools AND on a Linux 64 bit build of the tools.
Just to recap - on a clean installation (Eclipse, CDT, GME plugins etc.) Window > Preferences > C/C++ > Debug > GDB does not appear in the Preferences dialog.
After a GDB OpenOCD Debugging debug session is attempted (doesn't even have to succeed) then Window > Preferences > C/C++ > Debug > GDB does appear in the Preferences dialog.
It's like something in the act of trying a debug session changes some state allowing this option to appear in the Preferences.
I haven't tried to see if any other action (e.g. starting ANY debug session - not just GDB OpenOCD Debugging session) triggers this.
Do you not get the same behaviour on Mac?

Hope this helps.

ilg-ul, 2018-01-09:
I believe you are right, but the Debug -> GDB preferences page is managed by CDT, I don't how the plug-ins can affect it.

Could you try the same configuration, but without installing the GME plug-ins?

TM1234, 2018-01-09:
Good point.
Let me try that this afternoon and I'll post back asap.

TM1234, 2018-01-09:
OK - it happens when there is no GME at all so it's not a GME issue but rather an Eclipse or CDT issue.
I'll close this one and try to log one against the relevant Eclipse project.
Thanks.

closed 2018-01-09 by TM1234

TM1234, 2018-01-09:
Seems to be a known issue

https://www.eclipse.org/forums/index.php/t/1089461/

but I'm not sure if this is the bug report that covers it;

https://bugs.eclipse.org/bugs/show_bug.cgi?id=303440

(which hasn't been touched for c. 8 years now!) or if there's some other, more specific, bug(s).
I'll keep digging...