eclipse-embed-cdt/eclipse-plugins #194: linker fails with "cannot find -lg"

solderdev, 2017-04-07:
Hello!

I did a clean install of **Neon.3** with **gcc-arm-none-eabi-6-2017-q1** on **macos** sierra with java jdk 1.8.0_121. (all installation steps on the HP were followed. i was running the plugins in a mars environment before)

After creating an STM32F4 example project the build failed, because in the project properties - build - settings - toolchains tab, C++ compiler was set to g++ instead of c++ (the executable now seems to be renamed?!).

After that the build failed at the linker stage with:
`..../arm-none-eabi/bin/ld: cannot find -lg`

I didn't specify any custom settings - just loaded a template blinky project.

Also my old projects (mars environment) give the same linker error.

All the best,
Peter

ilg-ul, 2017-04-07:
I guess you have a wrong toolchain path. it should read `gcc-arm-none-eabi-6-2017-q1-update/bin`.

solderdev, 2017-04-07:
OK - i was able to circumvent the problem by checking "no startup or default libs (-nostdlib)" in the linker settings.

Now the linker works.

But the objcopy fails -.-
The path to objcopy is now different (not in the bin folder and not prefixed).
If i enter the absolute path in the "Command:" field it works.

ilg-ul, 2017-04-07:
you completely messed your configuration.

solderdev, 2017-04-07:
Did i? (didn't change anything - was a clean install .. and i'm not new to this..)

Did you try the [gcc-arm-none-eabi-6-2017-q1-update](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads)? It seems the folder structure changed..

ilg-ul, 2017-04-07:
> the folder structure changed

in what way? I have it opened in Finder and see no differences.

solderdev, 2017-04-07:
**gcc-arm-none-eabi-5_4-2016q3** user$ ls -la bin
total 64880
drwxr-xr-x@ 30 chibby  staff     1020 27 Sep  2016 .
drwxr-xr-x@  7 chibby  staff      238  7 Apr 09:29 ..
-rwxr-xr-x@  1 chibby  staff   912192 26 Sep  2016 arm-none-eabi-addr2line
-rwxr-xr-x@  2 chibby  staff   946408 26 Sep  2016 arm-none-eabi-ar
-rwxr-xr-x@  2 chibby  staff  1609620 26 Sep  2016 arm-none-eabi-as
-rwxr-xr-x@  2 chibby  staff   960408 26 Sep  2016 arm-none-eabi-c++
-rwxr-xr-x@  1 chibby  staff   907628 26 Sep  2016 arm-none-eabi-c++filt
-rwxr-xr-x@  1 chibby  staff   960336 26 Sep  2016 arm-none-eabi-cpp
-rwxr-xr-x@  1 chibby  staff    38612 26 Sep  2016 arm-none-eabi-elfedit
-rwxr-xr-x@  2 chibby  staff   960408 26 Sep  2016 arm-none-eabi-g++
-rwxr-xr-x@  2 chibby  staff   960240 26 Sep  2016 arm-none-eabi-gcc
-rwxr-xr-x@  2 chibby  staff   960240 26 Sep  2016 arm-none-eabi-gcc-5.4.1
-rwxr-xr-x@  1 chibby  staff    44024 26 Sep  2016 arm-none-eabi-gcc-ar
-rwxr-xr-x@  1 chibby  staff    44144 26 Sep  2016 arm-none-eabi-gcc-nm
-rwxr-xr-x@  1 chibby  staff    44152 26 Sep  2016 arm-none-eabi-gcc-ranlib
-rwxr-xr-x@  1 chibby  staff   518064 26 Sep  2016 arm-none-eabi-gcov
-rwxr-xr-x@  1 chibby  staff   479348 26 Sep  2016 arm-none-eabi-gcov-tool
-rwxr-xr-x@  1 chibby  staff  5574100 27 Sep  2016 arm-none-eabi-gdb
-rwxr-xr-x@  1 chibby  staff  5821460 27 Sep  2016 arm-none-eabi-gdb-py
-rwxr-xr-x@  1 chibby  staff   983228 26 Sep  2016 arm-none-eabi-gprof
-rwxr-xr-x@  4 chibby  staff  1280116 26 Sep  2016 arm-none-eabi-ld
-rwxr-xr-x@  4 chibby  staff  1280116 26 Sep  2016 arm-none-eabi-ld.bfd
-rwxr-xr-x@  2 chibby  staff   922192 26 Sep  2016 arm-none-eabi-nm
-rwxr-xr-x@  2 chibby  staff  1123744 26 Sep  2016 arm-none-eabi-objcopy
-rwxr-xr-x@  2 chibby  staff  1413864 26 Sep  2016 arm-none-eabi-objdump
-rwxr-xr-x@  2 chibby  staff   946424 26 Sep  2016 arm-none-eabi-ranlib
-rwxr-xr-x@  2 chibby  staff   517608 26 Sep  2016 arm-none-eabi-readelf
-rwxr-xr-x@  1 chibby  staff   912628 26 Sep  2016 arm-none-eabi-size
-rwxr-xr-x@  1 chibby  staff   912248 26 Sep  2016 arm-none-eabi-strings
-rwxr-xr-x@  2 chibby  staff  1123744 26 Sep  2016 arm-none-eabi-strip

**gcc-arm-none-eabi-6-2017-q1-update** user$ ls -la bin
total 45000
drwxr-xr-x  18 chibby  staff      612 17 Feb 00:24 .
drwxrwxrwx   7 chibby  staff      238  6 Apr 22:36 ..
-rwxr-xr-x   1 chibby  staff  1016744 16 Feb 19:32 arm-none-eabi-addr2line
-rwxr-xr-x   1 chibby  staff  1206868 16 Feb 22:26 arm-none-eabi-c++
-rwxr-xr-x   1 chibby  staff  1016284 16 Feb 19:32 arm-none-eabi-c++filt
-rwxr-xr-x   1 chibby  staff  1206788 16 Feb 22:26 arm-none-eabi-cpp
-rwxr-xr-x   1 chibby  staff    42648 16 Feb 19:32 arm-none-eabi-elfedit
-rwxr-xr-x@  1 chibby  staff  1202596 16 Feb 22:26 arm-none-eabi-gcc
-rwxr-xr-x   1 chibby  staff    44036 16 Feb 22:26 arm-none-eabi-gcc-ar
-rwxr-xr-x   1 chibby  staff    44132 16 Feb 22:26 arm-none-eabi-gcc-nm
-rwxr-xr-x   1 chibby  staff    44140 16 Feb 22:26 arm-none-eabi-gcc-ranlib
-rwxr-xr-x   1 chibby  staff   709900 16 Feb 22:26 arm-none-eabi-gcov
-rwxr-xr-x   1 chibby  staff   669700 16 Feb 22:26 arm-none-eabi-gcov-tool
-rwxr-xr-x   1 chibby  staff  6206716 17 Feb 00:19 arm-none-eabi-gdb
-rwxr-xr-x   1 chibby  staff  6467468 17 Feb 00:24 arm-none-eabi-gdb-py
-rwxr-xr-x   1 chibby  staff  1096504 16 Feb 19:32 arm-none-eabi-gprof
-rwxr-xr-x   1 chibby  staff  1017244 16 Feb 19:32 arm-none-eabi-size
-rwxr-xr-x   1 chibby  staff  1016800 16 Feb 19:32 arm-none-eabi-strings

ilg-ul, 2017-04-07:
```sh
ilg-mbp:gcc-arm-none-eabi-6-2017-q1-update ilg$ ls -l bin
total 73880
-r-xr-xr-x@ 1 ilg  staff  1016744 Feb 16 20:32 arm-none-eabi-addr2line
-r-xr-xr-x@ 2 ilg  staff  1051112 Feb 16 20:32 arm-none-eabi-ar
-r-xr-xr-x@ 2 ilg  staff  1749248 Feb 16 20:32 arm-none-eabi-as
-r-xr-xr-x@ 2 ilg  staff  1206868 Feb 16 23:26 arm-none-eabi-c++
-r-xr-xr-x@ 1 ilg  staff  1016284 Feb 16 20:32 arm-none-eabi-c++filt
-r-xr-xr-x@ 1 ilg  staff  1206788 Feb 16 23:26 arm-none-eabi-cpp
-r-xr-xr-x@ 1 ilg  staff    42648 Feb 16 20:32 arm-none-eabi-elfedit
-r-xr-xr-x@ 2 ilg  staff  1206868 Feb 16 23:26 arm-none-eabi-g++
-r-xr-xr-x@ 2 ilg  staff  1202596 Feb 16 23:26 arm-none-eabi-gcc
-r-xr-xr-x@ 2 ilg  staff  1202596 Feb 16 23:26 arm-none-eabi-gcc-6.3.1
-r-xr-xr-x@ 1 ilg  staff    44036 Feb 16 23:26 arm-none-eabi-gcc-ar
-r-xr-xr-x@ 1 ilg  staff    44132 Feb 16 23:26 arm-none-eabi-gcc-nm
-r-xr-xr-x@ 1 ilg  staff    44140 Feb 16 23:26 arm-none-eabi-gcc-ranlib
-r-xr-xr-x@ 1 ilg  staff   709900 Feb 16 23:26 arm-none-eabi-gcov
-r-xr-xr-x@ 1 ilg  staff   669700 Feb 16 23:26 arm-none-eabi-gcov-tool
-r-xr-xr-x@ 1 ilg  staff  6206716 Feb 17 01:19 arm-none-eabi-gdb
-r-xr-xr-x@ 1 ilg  staff  6467468 Feb 17 01:24 arm-none-eabi-gdb-py
-r-xr-xr-x@ 1 ilg  staff  1096504 Feb 16 20:32 arm-none-eabi-gprof
-r-xr-xr-x@ 4 ilg  staff  1438504 Feb 16 20:32 arm-none-eabi-ld
-r-xr-xr-x@ 4 ilg  staff  1438504 Feb 16 20:32 arm-none-eabi-ld.bfd
-r-xr-xr-x@ 2 ilg  staff  1035120 Feb 16 20:32 arm-none-eabi-nm
-r-xr-xr-x@ 2 ilg  staff  1241684 Feb 16 20:32 arm-none-eabi-objcopy
-r-xr-xr-x@ 2 ilg  staff  1536488 Feb 16 20:32 arm-none-eabi-objdump
-r-xr-xr-x@ 2 ilg  staff  1051112 Feb 16 20:32 arm-none-eabi-ranlib
-r-xr-xr-x@ 2 ilg  staff   564900 Feb 16 20:32 arm-none-eabi-readelf
-r-xr-xr-x@ 1 ilg  staff  1017244 Feb 16 20:32 arm-none-eabi-size
-r-xr-xr-x@ 1 ilg  staff  1016800 Feb 16 20:32 arm-none-eabi-strings
-r-xr-xr-x@ 2 ilg  staff  1241684 Feb 16 20:32 arm-none-eabi-strip
ilg-mbp:gcc-arm-none-eabi-6-2017-q1-update ilg$ 
```

solderdev, 2017-04-07:
um .. OK? that's strange .. i just re-downloaded the package from [LINK](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads) ([link to exact file](https://developer.arm.com/-/media/Files/downloads/gnu-rm/6_1-2017q1/gcc-arm-none-eabi-6-2017-q1-update-mac.tar.bz2?product=GNU%20ARM%20Embedded%20Toolchain,64-bit,,Mac%20OS%20X,6-2017-q1-update)).

and all the files are her .. LOL
in the archive i downloaded from the exact same page 30min ago the files are not there ... OMG

Build works now fine with original settings.
This might have been a mess-up on the download server?!

Thank you for the quick help!

ilg-ul, 2017-04-07:
> in the archive i downloaded from the exact same page 30min ago the files are not there

I doubt so.

do you have the habit of changing the files to r/o after installing from an archive? sometimes it might help.

solderdev, 2017-04-07:
Will do that.
Though it seems the 1st archive was broken (not sure why there was no error reported upon extraction). And it's quite strange i still got it to work with the above mentioned workarounds.

closed 2017-04-07 by ilg-ul