eclipse-embed-cdt/eclipse-plugins #195: Packs plug-in crashing on bad/nonexistent PDSC URL

haunma, 2017-04-07:
### Description

The following URL is returning a javascript-filled web page instead of a PDSC file:
 http://gd32mcu.21ic.com/data/documents/yingyongruanjian/GigaDeviceGD32F1x0_DFP.pdsc

This is causing my Eclipse installation to crash (unrecoverable except by killing the process) whenever I attempt to refresh the packs.

### Steps to Reproduce

1.  Refresh packs (as of 7 April 2017).

**Expected behavior:** Packs are refreshed, skipping any bad or un-parseable results from the list of pack URLs
**Actual behavior:** Eclipse crashes on bad PDSC URL

### Versions

* [plug-in version]  20170225
* [Eclipse version]  Neon.3 (4.6.3)
* [Java version]  Oracle JDK 1.8.0_121b13
* [operating system]  Windows 7
* [toolchain version]  GNU Tools for ARM Embedded, 6 2017-q1-update

ilg-ul, 2017-04-24:
unfortunately I cannot reproduce the problem. :-(

haunma, 2017-04-24:
Hmmm...  Is that because GigaDevice fixed their URLs?  Or they are still returning junk, and your installation doesn't crash on it?

ilg-ul, 2017-04-24:
I don't know, it looks like their URLs return correct XMLs.

please also try the new beta version from updates-test.

haunma, 2017-04-24:
I just tried the URL listed above in the issue.  It is still returning a script-filled web page with a "not found" message.  So if that URL is still in the pack list, then...  I don't know.  Maybe my problem was related to a bad Eclipse upgrade after all.

haunma, 2017-04-24:
If that URL is no longer in the pack list, and that is why you're not seeing the problem anymore, you could perhaps replicate the issue using the same trick that worked for me (only in reverse):  Copy the pack list to another web server, edit it to add the offending URL, and point Eclipse at the modified copy.

closed 2020-07-17 by ilg-ul