eclipse-embed-cdt/eclipse-plugins #197: Erroneous paths in index info?

TM1234, 2017-04-18:
Has anybody else noticed erroneous paths sometimes appearing in the index (?) info?
This does not seem to be predictable but sometimes I end up with entries like this in .metadata/.plugins//org.eclipse.cdt.core/<projectname>.xml. For example in one project just now I get (note the erroneous "path" in **bold**)

`<?xml version="1.0" encoding="UTF-8" standalone="no"?>
<project>
	<configuration id="ilg.gnuarmeclipse.managedbuild.cross.config.elf.debug.483978693" name="Debug">
		<extension point="org.eclipse.cdt.core.LanguageSettingsProvider">
			<provider id="ilg.gnuarmeclipse.managedbuild.cross.GCCBuiltinSpecsDetector">
				<language id="org.eclipse.cdt.core.gcc">
					**<entry kind="includePath" name="/${ProjName}/#undef __thumb2__">
						<flag value="BUILTIN|READONLY|VALUE_WORKSPACE_PATH"/>
					</entry>**
					<entry kind="includePath" name="C:/Microsemi/SoftConsole_v5.1/CMSIS/V4.5/Include">
						<flag value="BUILTIN|READONLY"/>
					</entry>
`
I am using Eclipse CDT Luna SR2 32 bit on Windows 7 64 bit with the GNU ARM Eclipse 3.2.1 plugins.
I don't recall seeing behaviour like this in earlier versions of the plugins.
Any ideas?
If you need any more info let me know.
Thanks,

TM1234, 2017-04-18:
OK - that didn't work very well - let me try to clean it up...

ilg-ul, 2017-04-18:
nope, I don't even know what is this file used for.

TM1234, 2017-04-18:
Apologies for the initial messed up post - I've cleaned it up now.
I don't know much about that file either but noticed that since I started using the 3.2.1 plugins that I sometimes get such erroneous "path" entries which appear as problems in the GUI.
I presume that this file is generated by the GNU ARM Eclipse plugins given the ilg.gnuarmeclipse.managedbuild... identifiers?
Unfortunately I can't identify a reproducible test case so far... :-(

ilg-ul, 2017-04-18:
> I presume that this file is generated by the GNU ARM Eclipse plugins given the ilg.gnuarmeclipse.managedbuild... identifiers?

not directly, maybe it is a CDT side effect, the file may be created by the indexer, and the wrong path may be the result of a parsing problem.

> Unfortunately I can't identify a reproducible test case so far... :-(

even if you can, you have to do it again on Neon.3; Luna is much too old to be of any value.

TM1234, 2017-05-01:
FWIW I see the same issue mentioned here:

http://stackoverflow.com/questions/42098200/strange-include-path-in-eclipse-cdt-gnu-arm

TM1234, 2017-06-07:
If it helps at all I can provide a Windows installer for a modified version of Microsemi SoftConsole (https://www.microsemi.com/products/fpga-soc/design-resources/design-software/softconsole) that installs all relevant tools (Eclipse Neon.3/CDT 9.2.1, GNU MCU Eclipse 4.6.3, GNU ARM Embedded Toolchain, RISC-V GNU Toolchain etc.) and when launched opens an example workspace containing a few Cortex-M1, Cortex-M3 and RISC-V (RV32IM) projects some or all of which, once indexed, will illustrate the problem.

ilg-ul, 2017-06-07:
I don't know what to say, windows is not exactly the platform of choice for me. :-(

TM1234, 2017-06-07:
VM?

ilg-ul, 2017-06-08:
yeah, very un-enthusiastically...

TM1234, 2017-06-08:
Well if there's anything that I can do at this end to diagnose/investigate let me know but I'm not really au fait with debugging Eclipse plugins unfortunately. But if I can do any tests or collate any info/logs...

Anyway if you do happen to be able to use a Windows VM then the installer that I referred to above is here: ftp://ftp.actel.com/incoming/tommy/SoftConsole_v5.2/Microsemi-SoftConsole-v5.2.0.1-Windows-Installer.exe. 

When installed and run it will open a default example workspace and often I get results like this:
[image]
Hope this helps.

ilg-ul, 2017-06-08:
I reopened the issue, in case there are other reports, but currently, given the low resources available, I need to prioritise things, and since Microsemi is not exactly a significant contributor to the project (http://gnuarmeclipse.github.io/finance/sponsorship/), it might take a while.

AntonKrug, 2017-11-15:
I replicated it on the Linux too, with our newest SoftConsole. I had a workspace where only one of the projects had this issue (probably run build previously there), closed the SC went to the workspace and by hand deleted all the indexes, this forced the CDT to reindex on next start and created these warnings for all projects present. I will try to look it when I will have spare time, but so far the suspects are our workspace, CDT and Eclipse MCU.

TM1234, 2018-01-15:
Re the "duplicate path entries found"  warning I can reproduce this if I change a project include path from, say, "../include" to "${workspace_loc:/${ProjName}/include}" (which is the format that the Edit directory path > Workspace... button generates), then build the project (which makes the warning go away if it was displayed), delete the build target folder (e.g. Debug), collapse the project in Project Explorer, exit Eclipse, run it again and expand the project in Project Explorer. 

For some reason when the include paths are specified in this format, using Eclipse variables to refer to the workspace and project, it causes this warning but I don't understand why. I was able to reproduce it with a GME STM32F4 template generated example project but I strongly suspect that this is a CDT rather than a GME issue. 

I haven't been able to identify a reproducible test case for the "Include path not found" warning mentioned above (and in the screenshot) so far.

Hope this helps.

TM1234, 2018-01-16:
BTW - while using relative paths seems to avoid the "duplicate path" warnings note that in Project Properties > C/C++ General > Paths and Symbols > Includes it says "Using relative paths is ambiguous and not recommended. It can cause unexpected effects".

ilg-ul, 2018-01-16:
>  I was able to reproduce it with a GME STM32F4 template generated example project but I strongly suspect that this is a CDT rather than a GME issue.

it is easy to test, create a native project, that does not use my plug-ins.

>  "Using relative paths is ambiguous and not recommended. It can cause unexpected effects" although I can't really see how...?

yeah, while working on the mbedOS exporter, I experimented with generating the .cproject file and when I tried to group the Debug/Release folders below a build folder, I had problems.

CDT really does not work if the build folders are moved in a different location.

TM1234, 2018-01-16:
> it is easy to test, create a native project, that does not use my plug-ins.

Of course.
I will try that tomorrow.
Thanks.

TM1234, 2018-01-16:
OK - so far I have been unable to reproduce the "Invalid project path: Duplicate path entries found" warning using Eclipse/CDT and the native Windows (MinGW) compiler or using the Arm GCC Embedded compiler (with "raw" CDT rather than GME plugins). I have only been able to reproduce it when using the GME plugins which surprises me...

FWIW I'm using:

- Windows 10 64 bit
- Eclipse 64 bit
- JRE 9.0.1 (and telling Eclipse to use this via eclipse.ini)
- Arm GCC Embedded 7-2017-q4-major 32 bit
- GME RISC-V GCC Tools 7.2.0-2
- GME Windows Build Tools 2.10 64 bit 
- GME 4.3.1 for the GME tests

AntonKrug, 2018-01-16:
I wonder if the resolve of macros has to do something with it, with -consoleLog I see entries from .cross and from .cross.riscv4 as well:

> !ENTRY ilg.gnumcueclipse.managedbuild.cross 4 1 2018-01-12 11:49:59.339
> !MESSAGE arm.PathEnvironmentVariable.resolveMacros("${eclipse_home}/../build_tools/bin:/opt/gcc-arm-none-eabi-4_9-2015q3/bin", "Release") = ""
> 
> !ENTRY ilg.gnumcueclipse.managedbuild.cross.riscv 4 1 2018-01-12 11:50:00.044
> !MESSAGE arm.PathEnvironmentVariable.resolveMacros("${eclipse_home}/../build_tools/bin:/opt/gcc-arm-none-eabi-4_9-2015q3/bin", "Debug") = ""

ilg-ul, 2018-01-16:
yes, processing macros is a big issue in CDT, it might be something related to them.

ilg-ul, 2018-01-16:
I suggest you enable debug for all plug-ins and check the console

https://gnu-mcu-eclipse.github.io/developer/eclipse/debugging-options/

TommyMurphyTM1234, 2020-12-10:
I guess that this could be closed at this stage?

ilg-ul, 2020-12-10:
Things changed significantly since then, so it might no longer apply.

I suggest you try the new 2020-12 release and open a new ticket if needed.

closed 2020-12-10 by ilg-ul

TommyMurphyTM1234, 2020-12-10:
Thanks Liviu - I intend to try the latest plugins as soon as I get a chance
Apologies for the off topic question here but what was the outcome of the migration with regard to backward compatibility with existing GNU MCU Eclipse projects? Compatible or incompatible?
Thanks

ilg-ul, 2020-12-10:
The intent was to preserve compatibility with existing projects while importing them in a new workspace.

Compatibility with existing workspaces is not guaranteed, since it might require resetting some perspectives, thus the recommendation to create a new workspace and import the projects there.

Compatibility with 3rd party plug-ins could not be preserved, since the package names and IDs had to be changed to match the Eclipse Foundation requirements.

Another change is the minimum Eclipse, which is now 2020-09.

So, for a while, I suggest you stick with the latest v5 for your production builds, and in the meantime update for the new v6.

TommyMurphyTM1234, 2020-12-10:
Thanks @ilg-ul - I'll hopefully get to play with the updated plugins soon.