eclipse-embed-cdt/eclipse-plugins #201: Strange build error ... bin/sh: syntax error: unexpected "("

gbmont, 2017-04-23:
Hello,

Today I tried to convert an old lpc2148 makefile driven project to the latest gnuarmeclipse tools. I used the latest plugins, with the latest build tools. As toolchain I also use gcc 6.1 from launchpad.
Most of the build seems to work as expected but on the linker step I get this error which I do not know how to fix.

```Shell
Building target: main.elf
Invoking: Cross ARM GNU C Linker
arm-none-eabi-gcc -mcpu=arm7tdmi -mthumb-interwork -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections .....  
c:/program files (x86)/gnu arm eclipse/build tools/2.8-201611221915/bin/sh: syntax error: unexpected "("
Finished building target: main.elf
``` 

So my problem ist the  line:
```shell
c:/program files (x86)/gnu arm eclipse/build tools/2.8-201611221915/bin/sh: syntax error: unexpected "("
```
When I tried to add and build a empty "Hello World Cross ARM project" the build works as expected. So I do not think it isn´t a toolchain or build tools issue. It is really strange for me. 

I use the a Eclipse Neon.3 and my target OS is WIN10 

many thanks
cheers
geri

ilg-ul, 2017-04-24:
it is hard to tell what is wrong.

but are you sure this is a bug? if not, for general support, please use the project forum.

johnmoule, 2017-04-24:
I've seen the same linker error when I've had preprocessor symbols which contain parenthesis. The solution is to surround the symbol with quotes.

For example if the following type of value is contained in project properties > C/C++ Build > Settings > Cross ARM C Compiler > Preprocessor > Defined Symbols:
  XYZ=((uint32_t)48000000)

then quote it thus:

  "XYZ=((uint32_t)48000000)"

ilg-ul, 2017-04-24:
good point! 

please copy/paste the full compile line here, for checking.

gbmont, 2017-04-24:
Hello together

Thank you for your fast response. Sorry, probably it is not a bug of the plugin!  Attached the lines

> 
23:49:39 **** Incremental Build of configuration Debug for project AKFW ****
make all 
Building file: ../src/main.c
Invoking: Cross ARM GNU C Compiler
arm-none-eabi-gcc -mcpu=arm7tdmi -marm -mthumb-interwork -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3 -D"GCC" -D"USE_USB" -D"WINARM" -D"LPCUSB" -D"AKDEBUG" -D"CODESOURCERY" -I"C:\privat\CNC\AKFW\Misc" -I"C:\privat\CNC\AKFW\LPCUsbStack" -I"C:\privat\CNC\AKFW\HW_LPC" -I"C:\privat\CNC\AKFW\Controller" -I"C:\privat\CNC\AKFW\APPTEST_LPCUSB2012" -I"C:\privat\CNC\AKFW\APPTEST_LPC\intermediate" -std=gnu11 -MMD -MP -MF"src/main.d" -MT"src/main.o" -c -o "src/main.o" "../src/main.c"
../src/main.c:86:4: warning: #warning "Compile APPTEST Debug" [-Wcpp]
   #warning "Compile APPTEST Debug"
    ^~~~~~~
Finished building: ../src/main.c
.....
..... snipped Building lines
.....
Building file: ../APPTEST_LPCUSB2012/usbhw_lpc.c
Invoking: Cross ARM GNU C Compiler
arm-none-eabi-gcc -mcpu=arm7tdmi -marm -mthumb-interwork -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3 -D"GCC" -D"USE_USB" -D"WINARM" -D"LPCUSB" -D"AKDEBUG" -D"CODESOURCERY" -I"C:\privat\CNC\AKFW\Misc" -I"C:\privat\CNC\AKFW\LPCUsbStack" -I"C:\privat\CNC\AKFW\HW_LPC" -I"C:\privat\CNC\AKFW\Controller" -I"C:\privat\CNC\AKFW\APPTEST_LPCUSB2012" -I"C:\privat\CNC\AKFW\APPTEST_LPC\intermediate" -std=gnu11 -MMD -MP -MF"APPTEST_LPCUSB2012/usbhw_lpc.d" -MT"APPTEST_LPCUSB2012/usbhw_lpc.o" -c -o "APPTEST_LPCUSB2012/usbhw_lpc.o" "../APPTEST_LPCUSB2012/usbhw_lpc.c"
Finished building: ../APPTEST_LPCUSB2012/usbhw_lpc.c
Building target: AKFW.elf
Invoking: Cross ARM GNU C Linker
arm-none-eabi-gcc -mcpu=arm7tdmi -marm -mthumb-interwork -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections  -g3 -T "C:\privat\CNC\AKFW\APPTEST_LPCUSB2012\lpc2148-rom.ld" -nostartfiles -Xlinker --gc-sections -Wl,-Map,"AKFW.map" --specs=rdimon.specs -Wl,--start-group -lgcc -lc -lc -lm -lrdimon -Wl,--end-group -o "AKFW.elf"  ./src/main.o  ./Misc/ExtIoSpi.o ./Misc/cncmath.o ./Misc/crc8.o ./Misc/misc.o ./Misc/uCmdBuf.o ./Misc/uCtrlBuf.o ./Misc/uJogBuf.o ./Misc/ucharbuf.o  ./LPCUsbStack/usbcontrol.o ./LPCUsbStack/usbhw_lpc.o ./LPCUsbStack/usbinit.o ./LPCUsbStack/usbstdreq.o  ./HW_LPC/LPC2148_SPI.o ./HW_LPC/LpcExtIo.o ./HW_LPC/LpcHardw.o ./HW_LPC/LpcSpindle.o ./HW_LPC/LpcUart.o ./HW_LPC/uart.o ./HW_LPC/uartISR.o  ./Controller/CalcSpeed4D.o ./Controller/ExtIo.o ./Controller/ManualStep.o ./Controller/uJog.o ./Controller/uMotion.o ./Controller/uVector.o ./Controller/upcnc.o  ./APPTEST_LPCUSB2012/LpcPhase.o ./APPTEST_LPCUSB2012/LpcUsb.o ./APPTEST_LPCUSB2012/Startup\ (2).o ./APPTEST_LPCUSB2012/armVIC.o ./APPTEST_LPCUSB2012/console.o ./APPTEST_LPCUSB2012/startup.o ./APPTEST_LPCUSB2012/usbhw_lpc.o   
c:/program files (x86)/gnu arm eclipse/build tools/2.8-201611221915/bin/sh: syntax error: unexpected "("
Finished building target: AKFW.elf
 
Invoking: Cross ARM GNU Create Flash Image
arm-none-eabi-objcopy -O ihex "AKFW.elf"  "AKFW.hex"
arm-none-eabi-objcopy: 'AKFW.elf': No such file
make: *** [AKFW.hex] Error 1
makefile:54: recipe for target 'AKFW.hex' failed

> 

My preprocessor under roject properties > C/C++ Build > Settings > Cross ARM C Compiler > Preprocessor > Defined Symbols
GCC
WINARM
CODESOURCERY
LPCUSB
AKDEBUG
USEUSB

I also tried to uses quotes like 
"GCC"
"WINARM"...

Best regards and thank you again
Geri

closed 2017-04-24 by gbmont