eclipse-embed-cdt/eclipse-plugins #202: Eclipse Neon and ESP32

rsnbusa, 2017-04-24:
The problem is Eclipse Neon Version: Neon.3 Release (4.6.3)
Build id: 20170314-1500 for OSX
OSX: 10.11.6

ESP32 DevKitC
JTAG Debugger Flyswatter2

"Manual" Scenario.
OpenOcd (from EspressIf) starts Ok
start xtensa-esp32-elf-gdb connects to OpenOcd
Blinky is Blinking
Break. Stops
C ontinues
b X set breakpoint
C breaks at set point
print level prints ok
list Ok
i threads list all threads (FreeRTOS)
thread 3 set that thread
etc
etc
Well, in other words it works perfectly.

Now with Eclipse, Debug configuration
Same openocd executabel
Same xtensa-esp32-elf-gdb debugger with set mem inaccessible-by-default off
on Startup the Load Executable is diabled (ESP32 cannot load from opencod still)
Set Breakpoint name is app_main

Console displays the loading of openocd shows the following:

Open On-Chip Debugger 0.10.0-dev-g90071eb (2017-04-20-17:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz (speed also tested at 200)
force hard breakpoints
Started by GNU ARM Eclipse
Info : clock speed 1000 kHz
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Target halted, pc=0x40000400
Info : esp32.cpu1: Target halted, pc=0x40000400
Info : accepting 'gdb' connection on tcp/3333
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32.cpu0: Target halted, pc=0x40000400
esp32.cpu0: target state: halted
Info : esp32.cpu1: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32.cpu1: Target halted, pc=0x40000400
esp32.cpu1: target state: halted
Info : Auto-detected RTOS: FreeRTOS
Info : JTAG tap: esp32.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32.cpu0: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32.cpu0: Target halted, pc=0x40000400
esp32.cpu0: target state: halted
Info : esp32.cpu1: Core was reset (pwrstat=0x1F, after clear 0x0F).
Info : esp32.cpu1: Target halted, pc=0x40000400
esp32.cpu1: target state: halted
===== Xtensa registers
(0) pc (/32): 0x40000400
a lot of registers (where is the command to do this?)

Result:

The app IS NOT blinking (target halted in theory)
Click on [debugconfig name] and shows the app the Eclipse Menu which does not show a Resume Button(gray) but the Suspend and Terminate buttons are active. Its supposed to be halted according to openocd.
Click Suspend
debugapp.elf	
	Thread #2 1073413424 (No Name : Running) (Suspended : Signal : SIGINT:Interrupt)	
		0x40000400	
	Thread #3 1073446248 (k_task) (Suspended : Container)	
	Thread #4 1073445364 (Svc) (Suspended : Container)	
	Thread #5 1073410676 (No Name) (Suspended : Container)	
	Thread #6 1073408876 (No Name) (Suspended : Container)	
	Thread #7 1073442944 (No Name) (Suspended : Container)	

Thread 3 is the app_main entry point.
IF I step into or over it crashes with 
Error: esp32.cpu0: esp108_write_dirty_registers (line 793): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: esp32.cpu0: xtensa_resume (line 873): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: esp32.cpu0: esp108_fetch_all_regs (line 649): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: esp32.cpu0: esp108_fetch_all_regs (line 673): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: Exception reading pc!

Error: esp32.cpu0: xtensa_step: Stepping doesn't seem to change PC!
Error: esp32.cpu0: esp108_write_dirty_registers (line 793): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: esp32.cpu0: esp108_fetch_all_regs (line 649): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: esp32.cpu0: esp108_fetch_all_regs (line 673): DSR (81A0CC13) indicates DIR instruction generated an exception!
Error: Exception reading pc!

If I dont do that and simple Resume, no Blinking, seems to be stuck in a loop at  0x40000400 WHICH IS NOT  the entry point.

Have tried everything, can u help.

Regards.

Robert

ilg-ul, 2017-04-25:
are you convinced that this is due to a bug in the GNU ARM Eclipse plug-ins?

for general support, please use the project forum.

closed 2017-04-25 by ilg-ul

rsnbusa, 2017-04-25:
Thanks for your reply.

The basis for this is that when I do the “manual” debugging, that is, load openocd (plain and simple) and then load xtensa-gdb (again without parameters) it works, it blinks, stops, resumes,print vars, shows threads, hits breakpoints, etc. When the same two apps are called from Eclipse it does not and is stuck at a point that is not part of the Blink app or the flashed code . Also, when using the manual option there is an X list of threads and when done via Eclipse its different.

Anyway, how can I proceed or where to get suggestions and help. I presume that you are stating that this request will not go any further and I have to look for other ways of solving it.

Regards.

ilg-ul, 2017-04-25:
http://gnuarmeclipse.github.io/support/forum/

and please do not reply by email, use the web interface.