eclipse-embed-cdt/eclipse-plugins #204: Can't see anything on Peripheral View: NXP KE02

RobertoPaolinelli, 2017-05-08:
### Description

[Description of the bug or feature]

Downloaded all packs. Installed only NXP KExx one (version 1.7.0)
Entered debug with P&E Multink FX, debug ok (memory is displayed) but Peripheral View does not show anything

### Versions

* [plug-in version]
GNU ARM C/C++ Packs 2.2.2.201704251808

* [Eclipse version]
Neon.3 (4.6.3)

* [Java version]
1.8.0.131

* [operating system]
Windows 10 64 bit

* [toolchain version]
GNU ARM Embedded Toolchan 6q1 2017

Best Regards
Roberto

[image]

ilg-ul, 2017-05-08:
> P&E Multink FX

I have no experience with this probe and/or the associated debug plug-in, most probably it has no support for peripheral registers. 

officially supported debug plug-ins are: J-Link, OpenOCD and QEMU.

closed 2017-05-08 by ilg-ul

RobertoPaolinelli, 2017-05-08:
Thank you very much
Best Regards
Roberto