eclipse-embed-cdt/eclipse-plugins #205: ARM CP15 registers read/write support

Menion2k, 2017-05-08:
Hi all
Is there any plan to add read/write support for the CP15 registers?
I know that it would require some extension of the CDT debugger (that I am not sure it can come from CDT since it is CPU core dependent) and also support from the JTAG backend (SEGGER support it), but it would be of a great help for a more complete ARM core control

ilg-ul, 2017-05-08:
Eclipse shows exactly the registers provided by GDB, nothing more, nothing less.

when using a GDB server, GDB shows exactly the registers provided by the GDB server.

so... I doubt there is anything the debug plug-ins can do for this.

Menion2k, 2017-05-08:
I know
In fact, it would require an extension of the CDT debugger to work in conjuction with the specific JTAG backend. In the SEGGER case it would require to send some "monitor" command from GDB to SEGGER GDB server