eclipse-embed-cdt/eclipse-plugins #207: Error when creating OpenOCD debug launch config when project selected

TM1234, 2017-06-07:
See here:

https://groups.google.com/a/groups.riscv.org/d/msg/sw-dev/4sC6Fi_5pyA/MXCiPSysAAAJ

Windows 7 64 bit
Eclipse Neon.3 (4.6.3) + CDT 9 (9.2.1) 32 bit
Java 8u131
GNU MCU Eclipse plugins for ARM and RISC-V v4.6.3-20170606-0936

Select a project in the Project Explorer.
Select the menu option Run > Debug Configurations.
Select GDB OpenOCD Debugging 
Create a new debug launch configuration
The following error is displayed:

[image]

If no project is selected in the Project Explorer then that error does not appear and the debug launch configuration is successfully created - albeit with many fields blank and needing to be filled in manually (which seems to be correct behaviour in the circumstances).

Partial log attached:

[gme.log.txt](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/1059479/gme.log.txt)

If any further info or testing is needed then please let me know.

Regards
Tommy

ilg-ul, 2017-06-07:
thank you.

if you allow me, one short comment: I know that selecting user ids are a personal matter, but not adding the full name can lead to lots of confusion.

TM1234, 2017-06-07:
I did try to change my "display name" on my profile (before and again just now) but it still stays at TM1234 so I don't really know how to do anything about that. Sorry.

ilg-ul, 2017-06-07:
[image]

TM1234, 2017-06-07:
Yes - I changed my Name there but it still uses TM1234.

ilg-ul, 2017-06-08:
Tommy, could you check version 4.1.1-201706082015 from

http://gnu-mcu-eclipse.sourceforge.io/v4-neon-updates-experimental

TM1234, 2017-06-08:
Did you mean GNU MCU C/C++ OpenOCD Debugging 4.1.5.201706082109?
[image]
With that version I no longer get any error when creating a debug launch configuration when there is a project selected in the Project Explorer.

I'm a bit confused by the version numbering. You announced the v4.6.3 GME plugins here:

https://groups.google.com/a/groups.riscv.org/d/msg/sw-dev/4sC6Fi_5pyA/vJdsedOFCQAJ

But this version is v4.1.5. Or is the versioning of all GME plugins together separate from the versioning of individual component plugins?

BTW - I think that it was reported before/elsewhere (can't find it mentioned in issue tracker though) but the Main > C/C++ Application field gets filled in with a relative path using a backslash rather than a forward slash which is a small barrier to portability (i.e. will not work on Linux/non Windows OSs) whereas a forward slash seems to be globally portable. Would it be possible to have forward slash used on Windows too? Will I open a separate issue for this?

Thanks a lot.

ilg-ul, 2017-06-08:
yes, the Eclipse versioning system is confusing, and will probably remain so.

for the GNU MCU Eclipse product, the version reflects the original version, like v4.6.3, followed by  the date (20170606-0936).

for the GNU MCU Eclipse plug-ins, the version reflects the version of the entire update site, which, unfortunately, does not show when you list the features, but is indeed the version used to publish the update site (https://bintray.com/gnu-mcu-eclipse/v4-neon-updates-experimental/p2)

each Eclipse bundle has its own version.

each Eclipse feature has its own version. 

one or more bundles are referred by one or more features. generally each feature has a main bundle, with a similar name, and the same version.

with all this chaos, at run-time, you can find the original repository version by searching for the ilg.gnu-mcu-eclipse.core plug-in, which artificially follows the version of the repository.

so, when you encounter a reference to GNU MCU Eclipse plug-ins version x.y.x-YYYYMMDDHHMM, the really relevant part is the date, which must be the same for all features, plug-ins, repository, etc. since it identifies the build time.

ilg-ul, 2017-06-08:
> Did you mean GNU MCU C/C++ OpenOCD Debugging 4.1.5.201706082109?

given the above explanation, "check version 4.1.1-201706082015" means get a version from the update site with a date later than 20170608-2015. since 20170608-2109 is later, yes, this a suitable version to test.

ilg-ul, 2017-06-08:
> the Main > C/C++ Application field gets filled in with a relative path using a backslash

as far as I know, that field is automatically filled in by CDT, not by the plug-in.

> Would it be possible to have forward slash used on Windows too? Will I open a separate issue for this?

if you know how to configure CDT to do this, yes, please open a separate issue and possibly provide a pull request.

TM1234, 2017-06-08:
Thanks for the explanations of the version numbering.
I wasn't aware that the backslash was coming from CDT rather than the GME plugin. I'll see if I can find anything on how this might be changed and if I find anything useful I will log an issue and/or PR.

ilg-ul, 2017-06-09:
I checked the code and indeed the TabMain used in the debugging plug-ins simply inherit from a parent CDT class.

Unfortunately that class is marked as deprecated, so at a certain moment it'll require attention.

Please log an issue, and, if you can find a solution how to reimplement this tab and rewrite the path to use forward slashes, please let me know. Being common to all debuggers, the new class must be located in ilg.gnumcueclipse.debug.gdbjtag, and instantiated in each debugging plug-in.

ilg-ul, 2017-07-11:
fixed in 4.1.1.201707111115

closed 2017-07-11 by ilg-ul