eclipse-embed-cdt/eclipse-plugins #208: Debug launch configuration Main > C/C++ Application field uses non portable backslash on Windows

TM1234, 2017-06-09:
See discussion of this issue towards the end of this issue #207 

On Windows a debug launch configuration created when there is a project selected in the Project Explorer will try to automatically fill out the Main > C/C++ Application field with a reference to the executable to be debugged (or the user can browse/search the project to select one if necessary).
On Windows a backslash is used - e.g. Debug\myproject.elf. 
On Linux/non Windows a forward slash is used - e.g. Debug/myproject.elf.
The backslash will not work on non Windows platforms.
The forward slash is portable across Windows and Linux/non Windows platforms.
As such it would be better if a forward slash was used for portability.
Otherwise the Windows user must manually edit the field to change backslash to forward slash if the debug launch configuration is to be portable to non Windows OSs.

As explained in #207 by Liviu this behaviour comes not from the GNU MCU Eclipse plugins but from CDT itself - and a deprecated class at that (TabMain).
It would be good to identify a fix/mod that would allow forward slashes to be used as standard regardless of the host OS.
At the moment I cannot propose a solution.

(I don't know if there are any other contexts/fields in which this might also be desirable?).

ilg-ul, 2017-06-09:
please note that in normal conditions the debug configurations are not stored in the project, but in the workspace. as such, project portability is not affected.

only when the debug configurations are explicitly saved in the project (Tab Common -> Save as -> Shared file) this becomes an issue.

the solution implies identifying the new class for the Main tab (to fix the deprecation warning) and creating a new class derived from it, with the additional behaviour of rewriting the path with the POSIX separators, then using this new class in the debugging plug-ins.

ilg-ul, 2017-06-09:
> I don't know if there are any other contexts/fields in which this might also be desirable?

all fields that, in one way or another, end up in the project, **must** be saved in a portable format. if you identify other such fields, please let me know.