eclipse-embed-cdt/eclipse-plugins #209: Tools Paths preference

jief666, 2017-06-15:
in Preferences -> C/C++ Build -> Tools Paths, if you enter ${ProjDirPath} in field "Build tools folder" or "Toolchain folder", eclipse says "Value must be an existing directory".

If you manually enter ${ProjDirPath} in "ilg.gnuarmeclipse.managedbuild.cross.prefs", it's WORKING !!! But then, if you click on the Peferences page again, you cannot leave it.

So, it's great it's working as I love self contained project. Would be nice not to have this error message, and even a textfield saying the resolved path (I saw that elsewhere in eclipse).

Thanks.

ilg-ul, 2017-06-15:
validating folders is the default Eclipse behaviour.

you can disable this with a preference (http://gnuarmeclipse.github.io/developer/preferences/ilg.gnuarmeclipse.managedbuild.cross/)

jief666, 2017-06-15:
Eclipse won't substitute ${ProjDirPath} ? If I enter "${workspace_loc}", the validation works...

Thanks for the workaround. Where should I set workspace.toolchain.path.strict=true ?

ilg-ul, 2017-06-15:
http://gnuarmeclipse.github.io/developer/eclipse/runtime-preferences/#defaults

jief666, 2017-06-15:
If it can help someone : 
  -pluginCustomization is followed by a path to an ini file
  - defaults have to be prefixed with ilg.gnuarmeclipse.managedbuild.cross/. Ex:ilg.gnuarmeclipse.managedbuild.cross/global.toolchain.path.strict=true

ilg-ul, 2017-06-15:
defaults have to be prefixed with the plug-in id. for the toolchain path, the id is indeed `ilg.gnuarmeclipse.managedbuild.cross`.

ilg-ul, 2017-06-15:
btw, keeping tools inside the Eclipse workspace is generally a bad idea.

I'm currently working on an separate package manager to install multi-version binary tools (toolchain, windows build tools, openocd, qemu, etc) in a central location, be default somewhere in the user home folder. multi-version means you can have multiple versions of the same tool installed at the same time.

jantje, 2017-06-15:
> I'm currently working on an separate package manager to install multi-version binary tools (toolchain, windows build tools, openocd, qemu, etc) in a central location, be default somewhere in the [...]

FYI This is exactly what Arduino does with their third party hardware and which is implemented in sloeber. 
I think Sloeber and gnu-mcu-eclipse are growing more and start overlapping. As you know the sloeber product already contains  gnu-mcu-eclipse plugins. I think it would be smart if we collaborated more.

jief666, 2017-06-15:
I know that many people like the "central location" for a lot of thing. Even eclipse with this oomph make some "central" cache of setup , update, installed plugins. I think everything is in .eclipse and .p2.
I, personally, hate that. Each time you move your workspace (another location, an external disk, another computer, etc.) there is something missing.
So I make self sufficient workspace with every tools I need in it (including eclipse). No absolute path at all. I can click on workspace I didn't open in years and everything compile.
For me, it's an illusion to believe that few years later, you still have the old gcc you need to compile an old project, located at the same path. At least for me, it never happened. Change of computer, reinstall of operating system, etc.
Also, I can download from my NAS on whatever computer I have and run without config.

So I say : keeping all the tools you need for a project inside the Eclipse workspace (or a sibling folder) is generally a GOOD idea.

jief666, 2017-06-15:
btw, I currently can do it with this version of the plugin. Please, don't remove that possibility.

closed 2017-06-15 by ilg-ul

ilg-ul, 2017-06-15:
> exactly what Arduino does with their third party hardware

is there any documentation on how this works?

> sloeber product already contains gnu-mcu-eclipse plugins. I think it would be smart if we collaborated more.

any suggestions?

jantje, 2017-06-15:
>is there any documentation on how this works?

There is this page https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification
However Arduino starts from boards. Boards are delivered by a platform.
From a toolchain perspective it is important to know platform=toolchain and toolchain commands (defined as recipes in platform.txt).
Platforms are described in json files. The json files point to archives that contain the tools for the toolchain. The manager needs to downloaded and install these tools for the platform to  be available.
Plenty of hardware providers have created a json file for their boards as you can see here https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls

In theory all these boards work with sloeber. My last test successfully did +400 boards.

How you implement this is up to the toolbuilder. Simplified Sloeber puts all info in boards.txt and platform.txt in environment variables. Sloeber also defines a toolchain based on the recipe environment variables. This makes it possible to swap the toolchain without changing the toolchain at the CDT level.

Note that it is perfectly possible to use the toolchains without using the arduino core. https://github.com/grbl/grbl is a nice example.

>any suggestions?

I think it would be good to have a look at each others tools. We both started from somewhere and grew over the years and now start overlapping. Things won't magically fit and the 2 entry points will remain but in the overlapping zone it is no use doing things twice.
I already shopped debugging with you.
Given the comments above it looks as if you are looking for a way to get many toolchains working quickly. If so Sloeber can help you out.

ilg-ul, 2017-06-16:
> it would be good to have a look at each others tools

right. I'll take a look at the Arduino links.

thank you,

Liviu