eclipse-embed-cdt/eclipse-plugins #212: Error displayed when using J-Link and SWD

akuhtz, 2017-06-20:
Hi,

I get an error when selecting SWD in the Debugger tab of the _GDB SEGGER J-Link Debugging_ configuration:
The reason is that in the launch configuration the following entry is generated with the interface in lowercase:
`
<stringAttribute key="ilg.gnuarmeclipse.debug.gdbjtag.jlink.gdbServerDebugInterface" value="swd"/>
`

Changing the value manually to uppercase makes the error disappear:
`
<stringAttribute key="ilg.gnuarmeclipse.debug.gdbjtag.jlink.gdbServerDebugInterface" value="SWD"/>
`

I'm using _SEGGER J-Link GDB Server V4.80_.

I think it comes from this definition:
https://github.com/gnu-mcu-eclipse/eclipse-plugins/blob/develop/bundles/ilg.gnumcueclipse.debug.gdbjtag.jlink/src/ilg/gnumcueclipse/debug/gdbjtag/jlink/preferences/DefaultPreferences.java#L25

... but mabe it's easier to just uppercase the value when it's written to the launch configuration:
https://github.com/gnu-mcu-eclipse/eclipse-plugins/blob/develop/bundles/ilg.gnumcueclipse.debug.gdbjtag.jlink/src/ilg/gnumcueclipse/debug/gdbjtag/jlink/Configuration.java#L121

ilg-ul, 2017-06-20:
> I get an error when selecting SWD in the Debugger tab

where do you see this message? can you copy it, or make a window capture?

the default interface is swd, so this setting is widely used, but I've never seen any error message related to it.

akuhtz, 2017-06-20:
When I start debugging (sorry I'm a newbie, can't describe better :-) ) I get the follwing error dialog:
[image]

When I change the value manually to (uppercase) SWD the error disappears.

akuhtz, 2017-06-21:
Hi Liviu,
I cannot reproduce it anymore :-( But I guess it was a problem with removing V6.10 and using V4.80 and not reboot.
I'll close this issue.

closed 2017-06-21 by akuhtz

ilg-ul, 2017-06-21:
ok, good luck with your projects!