eclipse-embed-cdt/eclipse-plugins #214: Should the plugin prevent the user from configuring illegal combinations of RISC-V architecture, integer abi, floating point abi etc.?

TM1234, 2017-06-30:
Should the GUI prevent the user from configuring illegal combinations of RISC-V architecture, abi and floating point etc.?
Or maybe it is deliberately permissive for simplicity and depends on the compiler tools to report any illegal options?
For example I can configure this illegal combination of options:

Architecture: RV32I (-march=rv32i)
Integer ABI: LP64 (-mabi=lp64)
Floating point ABI: Double precision (d)

I'm not sure if other Target Processor options might possibly need cross validation?

If it is considered acceptable for the GUI to be permissive and rely on the tools to report illegal combinations of command line flags then feel free to just close this issue. Thanks.

ilg-ul, 2017-06-30:
> Should the GUI prevent the user from configuring illegal combination

normally yes.

> deliberately permissive for simplicity

unfortunately the RISC-V specific options are too many/too complex, and it is very difficult to validate them in Eclipse.

simple buttons can be enabled/disabled acording to some conditions, but combo boxes cannot be restricted to certain values depending on other options, or it is very difficult to do it.

if you have concrete proposals for additional validation, please let me know, but I cannot guarantee the implementation is possible.

TM1234, 2017-06-30:
> if you have concrete proposals for additional validation, please let me know,

Unfortunately I don't know enough about the eclipse plugin GUI capabilities to offer specific suggestions but I wonder if there might be some common Eclipse plugin GUI mechanism whereby cross field validation can be done on the fly and if there are conflicting settings in force then some sort of warning icon/tooltip can be activated/displayed on the relevant field(s) and the Apply/OK etc. buttons disabled until the conflicts are resolved or cancelled? Maybe not given that I can't recall offhand any Eclipse plugin GUI working like this...

ilg-ul, 2017-06-30:
> buttons disabled until the conflicts are resolved

this is usual for other settings pages, but for the page with toolchain details I'm not sure it is possible, the current behaviour comes only from a large xml, not from java code.