eclipse-embed-cdt/eclipse-plugins #216: question about how to use Eclipse Headless Build

gilgameshskytrooper, 2017-07-10:
I am trying to create a script that can clean build a project I am working on in Eclipse.

The issue is with the command to run a clean build using Eclipse headless. This is the command I am using:

`Eclipse.app/Contents/MacOS/eclipse --launcher.suppressErrors -nosplash -data "/Users/andrewlee/git/SwingJS" -application org.eclipse.cdt.managedbuilder.core.headlessbuild -cleanBuild
`

This is essentially the structure of the command listed here [GNU ARM Eclipse](http://gnuarmeclipse.github.io/advanced/headless-builds/) (Looking at the main docs page of Eclipse is unhelpful as well).

The specific error message I am getting is as following:

```org.eclipse.logback.configuration: The org.eclipse.m2e.logback.configuration bundle was activated before the state location was initialized. Will try after the state location is initialized.
org.eclipse.m2e.logback.configuration: Logback config file: /Users/andrewlee/git/SwingJS/.metadata/.plugins/org.eclipse.m2e.logback.configuration/logback.1.7.0.20160603-1933.xml
SLF4J: Class path contains multiple SLF4J bindings.SLF4J: Found binding in [bundleresource://954.fwk1087712231:1/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [bundleresource://954.fwk1087712231:2/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html for an explanation.SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]org.eclipse.m2e.logback.configuration: Initializing logback
WARNING: No Project matched "swingjs". Skipping...
```

Now the paths I have in the command are correct as far as I know.

Project directory is not in ```.../workspace/SwingJS``` since I imported the project from Git using the Eclipse Git tool, and in that case, it keeps the project in ```$HOME/git/Project``` by default.

[image]
[image]

Furthermore, the Project name should be "swingjs", yet in the log, it says "WARNING: No Project matched "swingjs". Skipping..."

I would appreciate if someone who has actually successfully done this before gave me the specific commands they used to clean build an Eclipse project headlessly.

Thanks.

ilg-ul, 2017-07-10:
what is the bug you claimed you found?

gilgameshskytrooper, 2017-07-10:
No bug. I just got an automated email when I tried to email you.

Sorry

closed 2017-07-10 by ilg-ul

ilg-ul, 2017-07-10:
> an automated email when I tried to email you

and what did that email tell you?

when you registered the bug report, did you read the text in that page?