eclipse-embed-cdt/eclipse-plugins #218: Devices tab doesn't synchronize packs installed with cmsis-pack-eclipse plugin

engsergiosoares, 2017-07-13:
### Description

As discussed in the [forum](https://www.element14.com/community/message/225441/l/re-synchronize-devicesperipheral-view-with-cmsis-pack-installed-from-cmsis-pack-eclipse-plugin#225441), it seems the Packs plugin from GNU MCU Eclipse is not able to detect packs installed with the [cmsis-pack-eclipse](https://github.com/ARM-software/cmsis-pack-eclipse) plugin. Even after removing the cache file, it wasn't able to re-scan the installed packs. Both plugins were configured to point to the same location.

### Steps to Reproduce

1. Offline install a pack using the CMSIS-Pack manager [cmsis-pack-eclipse](https://github.com/ARM-software/cmsis-pack-eclipse) plugin
2. Create a new RTE project for this device, go to project properties and observe that the Device tab is empty.
3. Also, In the Packs tab (from the GNU MCU Eclipse not the CMSIS-Pack Manager tab), the list of installed packs is empty. 

**Expected behaviour:** Devices tab is updated regardless of which plugin installed a pack since they point to the same packs_path

**Actual behaviour:** Devices tab is always empty

### Versions

* Plug-in: ilg.gnuarmeclipse.repository-3.4.1-201704251808
* Eclipse Oxygen
* Windows 7
* GNU ARM Embedded 6

nalexopo, 2020-02-10:
I seem to have the same issue. I locally install a CMSIS pack which is visible in the package manager but I cant figure out how to start the project using the installed DFP.  Device is not present under Properties>C/C++ build > Settings>Devices. However if I install a Pack from the default keil site with the same procedure it is available. 

Since this is still Open can someone suggest a workaround?? Or a detailed guide on how to create a project using a locally installed DFP?

ilg-ul, 2020-02-10:
>  just allowing to manually set the SVD file in the Peripheral Registers View is enough

@engsergiosoares, BTW, I guess this feature was added some time ago.