eclipse-embed-cdt/eclipse-plugins #220: Unable to read repository at http://gnu-mcu-eclipse.netlify.com/

Eosis, 2017-07-18:
When trying to install the Eclipse tools from the website by dragging the install button across to the eclipse marketplace, I get an error saying that the repository cannot be read. When I try to wget the equivalent URL, I get a 404.

Is the service down or have I got some internal configuration error?

The specific URL I was trying is: https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/content.xml, and the output is below.

    wget https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/content.xml --no-check-certificate
    --2017-07-18 14:07:01--  https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/content.xml
    Resolving dl.bintray.com... 5.153.35.248
    Connecting to dl.bintray.com|5.153.35.248|:443... connected.
    HTTP request sent, awaiting response... 404 Not Found
    2017-07-18 14:07:01 ERROR 404: Not Found.

ilg-ul, 2017-07-18:
the server seems ok:

https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/

I checked to "Install new software" from it and it was ok.

I also drag & dropped the marketplace link to a test Eclipse and it opened the list of plug-ins.

Eosis, 2017-07-19:
So I have changed network now (moved from work to home network) and I'm seeing a slightly different issue.

Now, after dragging the install button across and hitting confirm, I get shown this dialog box.
[image]

Perhaps I should just use the pre-built eclipse binaries, instead of installing from the marketplace. This feels like it could be a problem with my systems.

Eosis, 2017-07-19:
A similar issue affected the subversive project and they seem to have traced it back to a package meta-data issue: https://www.eclipse.org/forums/index.php/t/796266/

ilg-ul, 2017-07-19:
indeed, I tried a fresh Oxygen with the marketplace button and something wrong happens.

but manually going to marketplace is ok. I updated the web page to reflect this.

ilg-ul, 2017-09-16:
duplicate of #228

closed 2017-09-16 by ilg-ul

ilg-ul, 2018-01-10:
Installs from Marketplace are now fixed.

AndrewGoedhart, 2018-05-24:
ou may also be hitting the Ubuntu 18.04 bug for java and SSL see

https://stackoverflow.com/questions/6784463/error-trustanchors-parameter-must-be-non-empty

trojandnc, 2018-05-31:
fyi: Had this issue today with on a fresh neon oxygen install. easy fix (for me) was to change protocol in URL from https to http.

http://gnu-mcu-eclipse.github.io/plugins/install/

profff, 2018-07-01:
Hi all 
I tried to install the plug-in on a fresh eclipse photon 
I tried by market place and by URL and I got every time the same issue : 
it find the pakage list correctly 
but when loading : 
it says : 
```
Unable to read repository at http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/content.xml.
java.lang.RuntimeException: Unexpected error: java.security.InvalidAlgorithmParameterException: the trustAnchors parameter must be non-empty
```

profff, 2018-07-02:
hi all 
sorry for disturbing I'm stuck in "trustanchors-parameter-must-be-non-empty" issue I tried to get back from openjdk 10 to oracle java 8 nothing helps ... :(

TM1234, 2018-07-02:
Did you look at the post from @AndrewGoedhart above and the link that he posted?

profff, 2018-07-02:
Y es I do I've also fresh up my ubuntu (intalled a 18.04 bare one) 
installed eclips photon c/c++
installed the Oracle JDK 8 ( doesnt let start eclipse photon )
Installed the default JDK witch is 
```
~$ java --version
openjdk 10.0.1 2018-04-17
OpenJDK Runtime Environment (build 10.0.1+10-Ubuntu-3ubuntu1)
OpenJDK 64-Bit Server VM (build 10.0.1+10-Ubuntu-3ubuntu1, mixed mode)
```
I fall into same issue 
I folowed @AndrewGoedhart 's link updated cacerts 
```
~$ /usr/bin/printf '\xfe\xed\xfe\xed\x00\x00\x00\x02\x00\x00\x00\x00\xe2\x68\x6e\x45\xfb\x43\xdf\xa4\xd9\x92\xdd\x41\xce\xb6\xb2\x1c\x63\x30\xd7\x92' > /etc/ssl/certs/java/cacerts
~$ /var/lib/dpkg/info/ca-certificates-java.postinst configure

```
I finally succeding in launching photon (but leads me with thousand error in glib and and few gtk warnings) anyway it starts I finally succedded in installing GNU MCU Eclipse plugin 4.3.3
I got a waring on authenticity for 
```
Warning: You are installing software that contains unsigned content. The authenticity or validity of this software cannot be established. Do you want to continue with the installation?
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.codered_1.1.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.codered.feature_1.1.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.core_4.3.3.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.core_1.2.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.gdbjtag_4.1.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.gdbjtag.jlink_4.3.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.debug.gdbjtag.jlink.feature_4.3.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.gdbjtag.openocd_4.3.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.debug.gdbjtag.openocd.feature_4.3.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.gdbjtag.pyocd_1.3.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.debug.gdbjtag.pyocd.feature_1.3.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.gdbjtag.qemu_3.3.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.debug.gdbjtag.qemu.feature_3.3.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.gdbjtag.restart_1.3.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.debug.packs_1.1.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.managedbuild.cross_1.2.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.managedbuild.cross.arm_2.6.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.managedbuild.cross.arm.feature_2.6.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.managedbuild.cross.riscv_2.8.1.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.managedbuild.cross.riscv.feature_2.8.1.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.managedbuild.packs_1.3.3.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.packs_1.3.1.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.packs.core_2.2.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.packs.data_2.2.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.packs.feature_2.2.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.packs.ui_1.3.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.ad_1.2.1.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.templates.ad.feature_1.1.5.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.core_2.6.1.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.cortexm_1.4.3.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.templates.cortexm.feature_1.4.3.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.freescale_2.2.9.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.templates.freescale.feature_2.2.9.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.freescale.pe_1.2.2.201804191501.jar
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.sifive_1.3.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.templates.sifive.feature_1.3.2.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/plugins/ilg.gnumcueclipse.templates.stm_2.6.4.201804191501
  /home/proff/ARMToolchain/eclipse_ARM/features/ilg.gnumcueclipse.templates.stm.feature_2.6.4.201804191501
```
I install anyway
Eclipse error handle somthing before reset : 
```
The following will be send to: Eclipse

REPORT

anonymousId         b6085cfa-a776-43c0-bd12-1735209333cc
name                proff
email               ogaste@free.fr
[... 190 lines cut ...]
version             1.3.500.v20171221-2204

name                org.eclipse.equinox.launcher
[... cut ...]
name                org.eclipse.jface
version             3.14.0.v20180423-0714

name                org.eclipse.swt
version             3.107.0.v20180611-0422

name                org.eclipse.ui
version             3.109.100.v20180426-0903

name                org.eclipse.ui.ide.application
version             1.2.100.v20180416-2050

name                org.eclipse.ui.ide
version             3.14.0.v20180517-1842

```
I 'll let you know as soon I succedding in cros compiling and cross debuging with my stlink v2

ilg-ul, 2018-07-02:
Apart from the warning for unsigned content, which is a known issue, I don't see other problems related to the GNU MCU Eclipse plug-ins. Do you?

profff, 2018-07-02:
no no it's ok (i've tried the workaround one time before reinstall all from scratch and it was not working but here with default JDK and the cacerts workarount it looks to works
no I have to figure out all the rest ! ;) 
thank's for the help

ilg-ul, 2018-08-17:
> SSL Certificates of these sites not being trusted by your http server.

I'm afraid I don't get it. what http server?

to me it looks like you have a problem with your java setup, which does not accept the certificates of the bintray.com server.

yoni64, 2018-10-24:
> There is one other reason that is contributing to this problem. SSL Certificates of these sites not being trusted by your http server. When you do an update, change the repository URL from https:// to http:// and continue to proceed with the installation. This solution has not been posted anywhere in any of the other forums. This issue is likely to happen on other packs that you try to install or register too.
> 
> For e.g. you may run into this issue when the site is: https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/
> 
> But you will succeed and get past this error if you change it to: http://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/
> 
> Aarthi Rishi.

There is a problem installing behind a proxy.
Replacing the address by this advice solve this problem. Thanks

ilg-ul, 2018-10-24:
sure, by using a http: address instead of a https: address you no longer need to trust the SSL certificate. 

but the problem remains, your setup has a problem with accessing some https servers.

ThomaSchneid, 2019-12-27:
Hi, 

I have issues when I try to install GNU MCU Eclipse from the marcetplace. I get the following message:

"Unable to read repository at http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/content.xml.
Remote host closed connection during handshake"

"javax.net.ssl.SSLHandshakeException: Remote host closed connection during handshake
	at com.sun.net.ssl.internal.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:882)
	at com.sun.net.ssl.internal.ssl.SSLSocketImpl.performInitialHandshake(SSLSocketImpl.java:1188)
	at com.sun.net.ssl.internal.ssl.SSLSocketImpl.writeRecord(SSLSocketImpl.java:654)
	at com.sun.net.ssl.internal.ssl.AppOutputStream.write(AppOutputStream.java:100)
	at java.io.BufferedOutputStream.flushBuffer(BufferedOutputStream.java:65)
	at java.io.BufferedOutputStream.flush(BufferedOutputStream.java:123)
	at org.apache.commons.httpclient.HttpConnection.flushRequestOutputStream(HttpConnection.java:828)
	at org.apache.commons.httpclient.MultiThreadedHttpConnectionManager$HttpConnectionAdapter.flushRequestOutputStream(MultiThreadedHttpConnectionManager.java:1565)
	at org.apache.commons.httpclient.HttpMethodBase.writeRequest(HttpMethodBase.java:2116)
	at org.apache.commons.httpclient.HttpMethodBase.execute(HttpMethodBase.java:1096)
	at org.apache.commons.httpclient.HttpMethodDirector.executeWithRetry(HttpMethodDirector.java:398)
	at org.apache.commons.httpclient.HttpMethodDirector.executeMethod(HttpMethodDirector.java:171)
	at org.apache.commons.httpclient.HttpClient.executeMethod(HttpClient.java:397)
	at org.apache.commons.httpclient.HttpClient.executeMethod(HttpClient.java:346)
	at org.eclipse.ecf.provider.filetransfer.httpclient.HttpClientFileSystemBrowser.runRequest(HttpClientFileSystemBrowser.java:227)
	at org.eclipse.ecf.provider.filetransfer.browse.AbstractFileSystemBrowser$DirectoryJob.run(AbstractFileSystemBrowser.java:69)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
Caused by: java.io.EOFException: SSL peer shut down incorrectly
	at com.sun.net.ssl.internal.ssl.InputRecord.read(InputRecord.java:462)
	at com.sun.net.ssl.internal.ssl.SSLSocketImpl.readRecord(SSLSocketImpl.java:863)
	... 16 more
"

I am totally stuck here

ilg-ul, 2019-12-28:
@ThomaSchneid, I cannot confirm this, on my machine I was able to install the plug-ins from netlify.

Java had many problems with ssl certificates, or with access via a proxy, or from a very isolated enterprise network.

I suggest you check the FAQs, known issues, etc.