eclipse-embed-cdt/eclipse-plugins #221: plugin for eclipse Oxygen version

bremenpl, 2017-07-18:
Hello there,
Are there any plans in the near future to make the plugin work in Oxygen version of eclipse?

ilg-ul, 2017-07-18:
when I conducted some tests, it was ok. 

why do you ask? did you have any problems?

bremenpl, 2017-07-18:
Yes, I installed the plugin by drag droping marketplace icon from the site. 
It said that my software ia not compatible but I clicked to install anyway. 
Then it asked either i want to update my envoroment to match this plugin, 
so i clicked yes. After restart there are no jlink options in the debug 
adding interface.

ilg-ul, 2017-07-18:
I double checked and Oxygen is checked as supported 

[image]

I don't know what went wrong.

bremenpl, 2017-07-18:
Ok, i will try again tommorow morning, but what installation method would 
you recomend?

ilg-ul, 2017-07-18:
for Oxygen? if marketplace fails, point the eclipse to the update site and install as usual.

bremenpl, 2017-07-19:
Hi,
Installing manually from help -> install new software worked without problems. Thanks.

ilg-ul, 2017-07-19:
installing from marketplace should work too, marketplace does nothing magic, it only provides the update site URL, as you can see in the image.

andrewleech, 2017-08-11:
I just had the same issue. Clean install of eclipse oxygen on windows 10.

Help -> Eclipse Marketplace. 
Search for "GNU MCU Eclipse' it comes up, version 4.1.1
Hit install, it starts to work then gives compatibility error, continue and it just updates a couple of eclipse built in plugins. 

Then went to Help -> Install New Software
GNU MCU Eclipse is in the sites drop down, select it, check the modules I want to install, go.
It works fine from there, restart and I've got the debug plugins working.

ilg-ul, 2017-08-11:
I'm sorry to hear that Marketplace install fails.

I'm afraid I have little to do to help.  :-(

EvetsMostel, 2017-09-08:
I get the same issue when trying to install from the marketplace.  I first get a message that states "The GNU MCU Eclipse 4.1.1 solution is not available.  Do you want to proceed anyway?"  When I click yes, I get a error message saying the "provisioning cannot be completed" and in the error log I get:

The following solutions are not available: GNU MCU Eclipse 4.1.1 (id=ilg.gnumcueclipse.debug.gdbjtag.jlink.feature.group, ilg.gnumcueclipse.debug.gdbjtag.qemu.feature.group, ilg.gnumcueclipse.packs.feature.group, ilg.gnumcueclipse.managedbuild.cross.arm.feature.group, ilg.gnumcueclipse.managedbuild.cross.riscv.feature.group, ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.group, ilg.gnumcueclipse.templates.stm.feature.group, site=http://gnu-mcu-eclipse.netlify.com/v4-neon-updates)

with:
eclipse.buildId=4.7.0.I20170612-0950
java.version=1.8.0_102
java.vendor=Oracle Corporation
BootLoader constants: OS=win32, ARCH=x86_64, WS=win32, NL=en_US
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os win32 -ws win32 -arch x86_64 -product org.eclipse.epp.package.cpp.product

ilg-ul, 2017-09-08:
@EvetsMostel, thank you for the details. 

did you manage to install directly from the update url?

ilg-ul, 2017-09-16:
duplicate of #228

closed 2017-09-16 by ilg-ul

soleilankerbaril, 2017-11-02:
hey i went into the help->install new software and cant seem to find GNU MCU eclipse

soleilankerbaril, 2017-11-02:
never mind i found it

ilg-ul, 2018-05-17:
I have no idea why Eclipse throws that error.

It is true that `content.xml` is not present in the update site, but this is the expected behaviour, since the compressed `content.xml.xz` is used.

You can see this at the actual URL:

- http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/

My current guess is a SSL issue.

AguirreTWoG, 2018-05-17:
Thank you guys for the fast response. I've just come back to my desk and tried tethering with my phone. I got  this this time:
_Unknown Host: http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/content.xml
gnu-mcu-eclipse.netlify.com_
I am going to try with the zips...

ilg-ul, 2018-05-17:
> tried tethering with my phone. .. Unknown Host: ...

I tried again on my mac and my Eclipse successfully reached the update site.

I would say that you have a problem with your machine, Java or network related.

AguirreTWoG, 2018-05-17:
OK. I'll check it. This is a new install I did last week and so far I have not gotten any other problem. What openjdk version does oxygen require? And thanks again for your help.

TM1234, 2018-05-17:
You should probably use Oracle Java rather than OpenJDK.
See here:

https://gnu-mcu-eclipse.github.io/plugins/install/#java

ilg-ul, 2018-05-18:
There seems to be a problem with Java & certificates:

https://gist.github.com/mikaelhg/527204e746984cf9a33f7910bb8b4cb6

There are more details on stackoverflow, but the reported problem is very old, it just resurfaced now:

https://stackoverflow.com/questions/6784463/error-trustanchors-parameter-must-be-non-empty/25188331#25188331

TM1234, 2018-05-18:
FWIW I just did a test and it worked fine for me:

1. Downloaded and extracted the Win64 Eclipse Oxygen 3.a + CDT package

1. Downloaded and extracted the Win64 Oracle JRE 10.0.1 into eclipse/jre

1. Ran Eclipse (which picked up the Java 10 in the eclipse/jre folder automatically).

1. Used Install New Software to point at http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/ and selected all GME plugins for installation and installation process went fine.

ilg-ul, 2018-05-18:
The problem seems specific to Ubuntu 18 and OpenJDK 10.

TM1234, 2018-05-18:
OK - sorry - I missed that detail.

ilg-ul, 2018-05-18:
I confirm that Ubuntu 18.04 and OpenJDK 10.0.1 do not work.

The solution is to install Oracle JDK.

A possible workaround is to recreate the certificates:

```
# 1. Save an empty JKS file with the default 'changeit' password for Java cacerts.
#    Use 'printf' instead of 'echo' for Dockerfile RUN compatibility.
/usr/bin/printf '\xfe\xed\xfe\xed\x00\x00\x00\x02\x00\x00\x00\x00\xe2\x68\x6e\x45\xfb\x43\xdf\xa4\xd9\x92\xdd\x41\xce\xb6\xb2\x1c\x63\x30\xd7\x92' > /etc/ssl/certs/java/cacerts

# 2. Re-add all the CA certs into the previously empty file.
/var/lib/dpkg/info/ca-certificates-java.postinst configure
```

For more details, see: https://gist.github.com/mikaelhg/527204e746984cf9a33f7910bb8b4cb6

AguirreTWoG, 2018-05-18:
Last night I continued looking into this and had found a thread with the suggestion:

_sudo update-ca-certificates -f 
sudo /var/lib/dpkg/info/ca-certificates-java.postinst configure_ 

But it did not work. I will see to use instead your suggestion and will report on the outcome. And, yes, I am using Kubuntu 18.04 with Java 10.0.1:

_openjdk version "10.0.1" 2018-04-17
OpenJDK Runtime Environment (build 10.0.1+10-Ubuntu-3ubuntu1)
OpenJDK 64-Bit Server VM (build 10.0.1+10-Ubuntu-3ubuntu1, mixed mode)_

ilg-ul, 2018-05-18:
> But it did not work

you need to first run the printf command before recreating the certificates.

the safe solution is to use Oracle Java.

AguirreTWoG, 2018-05-18:
Your suggestion worked. Thank you so much. I did not need to install the Oracle version of Java though.
To put it all in one place:

**$ sudo su
\# /usr/bin/printf '\xfe\xed\xfe\xed\x00\x00\x00\x02\x00\x00\x00\x00\xe2\x68\x6e\x45\xfb\x43\xdf\xa4\xd9\x92\xdd\x41\xce\xb6\xb2\x1c\x63\x30\xd7\x92' > /etc/ssl/certs/java/cacerts
\# /var/lib/dpkg/info/ca-certificates-java.postinst configure**

And the details of the installation:
Distro:
[image]

eclipse:
Eclipse IDE for C/C++ Developers
Version: Oxygen.3a Release (4.7.3a)
Build id: 20180405-1200
OS: Linux, v.4.15.0-20-generic, x86_64 / gtk 3.22.30

Java:
openjdk version "10.0.1" 2018-04-17
OpenJDK Runtime Environment (build 10.0.1+10-Ubuntu-3ubuntu1)
OpenJDK 64-Bit Server VM (build 10.0.1+10-Ubuntu-3ubuntu1, mixed mode)

ilg-ul, 2018-05-18:
great!

AguirreTWoG, 2018-05-18:
You are probably right. When I initially tried "sudo update-ca-certificates -f" it did not work, and so I had to open a root terminal. Most likely the final way will just do fine with sudo. I have just edited.