eclipse-embed-cdt/eclipse-plugins #225: PyOCD Debug Plugin Uses Incorrect Case For Override Targets

kellyjp, 2017-07-27:
### Description

When using the PyOCD debugging plug in with an unknown board it is necessary to override the target device as PyOCD cannot figure this out automatically.  The target choices provided for the override target field in the debugger tab of the PyOCD debug configurations window are all in upper case, but the PyOCD GDB server expects the target to be specified in lower case with the result that the PyOCD GDB server fails to start.  Manually overtyping in the override target field the desired target in lower case will result in the PyOCD GDB server successfully starting.

### Steps to Reproduce

Connect an unknown DAPLINK board to the computer (I'm using a Red Bear Blend 2).

1. Open the C/C++ perspective
2. Select Run -> Debug Configurations ... from the menu bar
3. Create a new "GDB PyOCD Debugging" configuration
4. Select the Debugger tab
5. Push the refresh button next to the board field and select the unknown board.
6. Select the tick box next to the override target field 
7. Select the target device for the unknown board in the override target field list box.  Note that all the target device names are in upper case (I selected NRF52).
8. Select the Apply button and then the Debug button

**Expected behaviour:**

The PyOCD GDB server starts and you are prompted to open the debug perspective (if not already open).

**Actual behaviour:**

A problem occurred message box is displayed stating that the pyOCD GDB server failed with code (2).  If you look at the GDB PyOCD Debugging console you see a usage message for PyOCD GDB server along with the error message: "pyocd-gdbserver: error: argument -t/--target: invalid choice" along with a list of valid choices all in lower case (including nrf52).

If the plugin is querying pyocd-gdbserver for supported targets (which it appears to do) it is somehow do a case conversion somewhere along the way that it shouldn't do.

### Versions

* GNU MCU C/C++ PyOCD Debugging Plug In Version 1.1.4.201707111115
* Eclipse Version 4.7.0.v2017061201800
* Java Version 1.8.0_131
* MacOS Version 10.12.6
* pyocd-gdbserver Version 0.8.1a1

The only issue I could find that is related to this is issue #82 - the fix for that issue seems to be in the code so I expect this is something different.

kellyjp, 2017-07-27:
I did a bit more analysis of this issue and I think I have the root cause.  When the pyocd-gdbserver is requested to list available targets in JSON by the Eclipse plugin it returns two strings for each target, one in upper case (part_number) and one in lower case (name).  Here is an example:

```
$ pyocd-gdbserver --list-targets --json
{
    "status": 0, 
    "version": {
        "major": 1, 
        "minor": 0
    }, 
    "pyocd_version": "0.8.1a1", 
    "targets": [
        {
            "part_number": "KL25Z", 
            "name": "kl25z"
        }, 
        {
            "part_number": "KL26Z", 
            "name": "kl26z"

....
```
The plugin is currently using the part_number when building the list of supported targets, and then passes the selected part_number (in upper case) when involving pyocd-gdbserver (which fails).  

A possible fix would be to change the plugin to use the name when building the list of supported targets instead of the part_number.

I'm not a Java person but here is a possible fix:

```
--- a/bundles/ilg.gnumcueclipse.debug.gdbjtag.pyocd/src/ilg/gnumcueclipse/debug/gdbjtag/pyocd/ui/TabDebugger.java
+++ b/bundles/ilg.gnumcueclipse.debug.gdbjtag.pyocd/src/ilg/gnumcueclipse/debug/gdbjtag/pyocd/ui/TabDebugger.java
@@ -1046,7 +1046,7 @@ public class TabDebugger extends AbstractLaunchConfigurationTab {
 
                        final ArrayList<String> itemList = new ArrayList<String>();
                        for (PyOCD.Target target : targets) {
-                               itemList.add(String.format("%s", target.fPartNumber));
+                               itemList.add(String.format("%s", target.fName));
                        }
                        String[] items = itemList.toArray(new String[itemList.size()]);
```

flit, 2017-07-30:
Hi @kellyjp. I'm the pyOCD plugin owner. Thanks for the detailed report! I'll take a look at fixing this shortly.

ilg-ul, 2017-11-05:
@flit, any progress?

I plan for a new release in about one week, so if you have a fix, please let me know.

flit, 2017-11-05:
Hi @ilg-ul, sorry, I probably won't have a fix in time for your next release. I have to reinstall the plugin development environment first, for some reason it stopped working correctly when you made to change from ARM to MCU. I was planning on doing this and fixing bugs while I'm on vacation in December.

ilg-ul, 2017-11-05:
ok.

ilg-ul, 2018-01-05:
@flit, any progress?

kellyjp, 2018-12-15:
@flit,

Thank you for fixing this issue !

James

ilg-ul, 2018-12-15:
A v4.5.1 beta release  which includes these patches can be installed from

http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

Please test it and let me know if everything is ok, since my test environment does not include pyOCD.

kellyjp, 2018-12-17:
@ilg-ul,

I installed the v4.5.1 beta from the link provided above and can confirm that when I select `NRF52` as the device in the debug configuration, pyocd-gdbserver is subsequently launched using the `--target nrf52` option i.e. the chosen device is correctly mapped into a pyocd target with the correct case.

James

ilg-ul, 2018-12-17:
thank you, James, for taking the time to test. 

if @flit has no further patches, this is it for now. it'll be out after 2018-12 goes public.

ilg-ul, 2019-01-01:
fixed since v4.5.1-201901011632

commit 7b75050c (2019-01-01): Fix pyOCD target override handling. (#329)

- TabDebugger maintains two maps to convert between target name and
  target part number. (Currently the part number is almost always just
  an uppercase version of the name, except for cortex_m/CoreSightTarget.)
- The target name saved in the configuration is converted to the part
  number for display in the target override combo box. The reverse
  conversion is applied when applying changes to the config.
- Unknown target names are supported by saving/displaying as-is.
- Also made the target name combo box 3x wider.
- This fixes #225.