eclipse-embed-cdt/eclipse-plugins #228: Installing from Marketplace fails with "The following solutions are not available: GNU MCU Eclipse 4.1.1"

amq, 2017-08-11:
### Description

Can't install the plugins from the Marketplace with the following error:

```
The following solutions are not available: GNU MCU Eclipse 4.1.1 (id=ilg.gnumcueclipse.debug.gdbjtag.jlink.feature.group, ilg.gnumcueclipse.debug.gdbjtag.qemu.feature.group, ilg.gnumcueclipse.packs.feature.group, ilg.gnumcueclipse.managedbuild.cross.arm.feature.group, ilg.gnumcueclipse.managedbuild.cross.riscv.feature.group, ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.group, ilg.gnumcueclipse.templates.stm.feature.group, site=http://gnu-mcu-eclipse.netlify.com/v4-neon-updates)
```

### Steps to Reproduce

1. Install Eclipse Neon.3 from scratch
2. Try installing the GNU MCU plugins from the Marketplace
3. Try opening http://gnu-mcu-eclipse.netlify.com/v4-neon-updates in browser to verify it's not some IP blacklisting etc, should open fine

**Expected behaviour:** Can install the plugins.

**Actual behaviour:** Got an error.

### Versions

* Eclipse version: Neon.3 Release (4.6.3)
* Java version: 1.8r131
* OS: Windows 10 and Ubuntu 16.04

katodo, 2017-08-31:
There is a solution or workaround to this bug?

ilg-ul, 2017-08-31:
yes, install directly from the url, as usual.

https://gnu-mcu-eclipse.github.io/plugins/install/

TM1234, 2017-09-15:
I just hit this issue too today.
I presume that this report is effectively a duplicate of this one?

https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/221

ilg-ul, 2017-09-15:
I guess so.

as long as installing from the URL is ok, there isn't much I can do. the problem is probably in the marketplace code.

TM1234, 2017-09-15:
I understand - installing from the update site rather than the marketplace is an easy workaround.
This post might also be a duplicate just in case it helps to close off duplicates and keep the issues tracker clean:

https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/220

ilg-ul, 2017-09-16:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=522394

ilg-ul, 2018-01-10:
It seems like the problem was in the Marketplace page defining the project (the feature IDs did not include the .feature at the end).

I fixed the definitions and now I was able to fully install from marketplace.

Please confirm.

TM1234, 2018-01-10:
I did a quick test on Windows 64 bit and it seems to work:

JRE 9.0.1
Eclipse Platform Runtime Binary
Configured eclipse.ini to point to "local" JRE
Ran it
Installed Marketplace client
Installed CDT from Marketplace
Installed GME plugins from Marketplace.
Created a SiFive example project

Hope this helps.

ilg-ul, 2018-01-10:
Thank you, then we can close this issue.

closed 2018-01-10 by ilg-ul