eclipse-embed-cdt/eclipse-plugins #231: Plugin build instructions are a little bit unclear

TM1234, 2017-09-04:
Just logging this for the record - not critical/urgent.

The plugin build instructions here are a little bit unclear and inconsistent:

https://gnu-mcu-eclipse.github.io/developer/build-procedure/
https://github.com/gnu-mcu-eclipse/eclipse-plugins

Specifically:

#1 - One set of instructions deals with a command line build while the other seems to be referring to using something (Eclipse?) to do the build. It might be helpful to document one or both build approaches more clearly?

#2 - https://gnu-mcu-eclipse.github.io/developer/build-procedure/

> The build result is in the ilg.gnumcueclipse.repository project
>
> the target/ilg.gnumcueclipse.repository-3.3.1-SNAPSHOT.zip file
> the target/repository folder, with the p2 update site

This is slightly out of date and a little confusing. It might be clearer if it was documented as something like:

`<build-root-folder>/gnu-mcu-eclipse/eclipse-plugins.git/repositories`

And there are actually two sub-folders and zips - not one

```
<build-root-folder>/gnu-mcu-eclipse/eclipse-plugins.git/repositories/ilg.gnumcueclipse.repository/target/ilg.gnumcueclipse.repository-x.y.z-SNAPSHOT.zip
<build-root-folder>/gnu-mcu-eclipse/eclipse-plugins.git/repositories/ilg.gnumcueclipse.riscv.repository/ilg.gnumcueclipse.riscv.repository-x.y.z-SNAPSHOT.zip
```
I presume that these zips can be used with Eclipse's Add New Software > Archive option but it's not clear from the documentation.

Hope this helps.

ilg-ul, 2017-09-04:
thank you, I'll take a look.

it looks like we have to deal with two issues, one being the accuracy of the README and the other being related to the separate documentation project, and both must be in sync.

ilg-ul, 2017-09-25:
I updated the README and the web pages.

if you notice other inconsistencies, please let me know.

ilg-ul, 2018-01-05:
I guess this was already fixed.

closed 2018-01-05 by ilg-ul