eclipse-embed-cdt/eclipse-plugins #235: An internal error occurred during: "Starting OpenOCD". named capturing group is missing trailing '}'

TM1234, 2017-09-13:
Anybody seen this error when trying to launch a debug session?

> Multiple problems have occurred during: "Starting OpenOCD".
> 
> Launching smartfusion2-cortex-m3-blinky Debug
> Starting OpenOCD
> 
> An internal error occurred during: "Starting OpenOCD".
> named capturing group is missing trailing '}'

My setup is...

> Eclipse Oxygen 32 bit on Windows 10 64 bit.
> 
> GNU MCU Eclipse plugins installed:
> 
>   GNU MCU C/C++ ARM Cross Compiler	2.5.1.201706030906	ilg.gnumcueclipse.managedbuild.cross.arm.feature.feature.group	Liviu Ionescu
>   GNU MCU C/C++ OpenOCD Debugging	4.1.5.201706030906	ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.feature.group	Liviu Ionescu
>   GNU MCU C/C++ RISC-V Cross Compiler	2.5.1.201706030906	ilg.gnumcueclipse.managedbuild.cross.riscv.feature.feature.group	Liviu Ionescu 

And I have created the projects and debug launch configurations from scratch within this environment and have not imported them from any previous version of the tools.

I can't see anything obviously wrong with my debug launch configuration - I haven't changed much from the default settings other than to specify my specific OpenOCD options (--command "set DEVICE M2S090" --file board/microsemi-cortex-m3.cfg, disable the Pre-run/Restart reset because it does not suit our targets and saved the debug launch configuration in the local project rather than in the workspace.

If you need any further details let me know.
Thanks.

TM1234, 2017-09-13:
Not sure if it helps at all but the workspace log has this:

```
!ENTRY org.eclipse.ui 4 0 2017-09-13 23:58:39.751
!MESSAGE Unhandled event loop exception
!STACK 0
java.lang.IllegalArgumentException: named capturing group is missing trailing '}'
	at java.util.regex.Matcher.appendReplacement(Matcher.java:841)
	at java.util.regex.Matcher.replaceAll(Matcher.java:955)
	at java.lang.String.replaceAll(String.java:2223)
	at ilg.gnumcueclipse.debug.gdbjtag.openocd.DynamicVariableResolver.resolveAll(DynamicVariableResolver.java:38)
	at ilg.gnumcueclipse.debug.gdbjtag.openocd.Configuration.getGdbServerCommand(Configuration.java:53)
	at ilg.gnumcueclipse.debug.gdbjtag.openocd.ui.TabDebugger.updateActualpath(TabDebugger.java:635)
	at ilg.gnumcueclipse.debug.gdbjtag.openocd.ui.TabDebugger.access$5(TabDebugger.java:632)
	at ilg.gnumcueclipse.debug.gdbjtag.openocd.ui.TabDebugger$6.modifyText(TabDebugger.java:385)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:180)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4428)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1079)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1103)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1084)
	at org.eclipse.swt.widgets.Text.wmCommandChild(Text.java:3122)
	at org.eclipse.swt.widgets.Control.WM_COMMAND(Control.java:4991)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4846)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5178)
	at org.eclipse.swt.internal.win32.OS.CallWindowProcW(Native Method)
	at org.eclipse.swt.internal.win32.OS.CallWindowProc(OS.java:2454)
	at org.eclipse.swt.widgets.Text.callWindowProc(Text.java:262)
	at org.eclipse.swt.widgets.Control.windowProc(Control.java:4941)
	at org.eclipse.swt.widgets.Text.windowProc(Text.java:2704)
	at org.eclipse.swt.widgets.Display.windowProc(Display.java:5178)
	at org.eclipse.swt.internal.win32.OS.SetWindowTextW(Native Method)
	at org.eclipse.swt.internal.win32.OS.SetWindowText(OS.java:3484)
	at org.eclipse.swt.widgets.Text.setText(Text.java:2323)
	at ilg.gnumcueclipse.debug.gdbjtag.openocd.ui.TabDebugger.initializeFrom(TabDebugger.java:691)
	at org.eclipse.debug.ui.AbstractLaunchConfigurationTabGroup.initializeFrom(AbstractLaunchConfigurationTabGroup.java:90)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupWrapper.initializeFrom(LaunchConfigurationTabGroupWrapper.java:201)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupViewer.displayInstanceTabs(LaunchConfigurationTabGroupViewer.java:766)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupViewer$8.run(LaunchConfigurationTabGroupViewer.java:637)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupViewer.inputChanged(LaunchConfigurationTabGroupViewer.java:654)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupViewer.setInput0(LaunchConfigurationTabGroupViewer.java:615)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationTabGroupViewer.setInput(LaunchConfigurationTabGroupViewer.java:590)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog.handleLaunchConfigurationSelectionChanged(LaunchConfigurationsDialog.java:1008)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog$4.selectionChanged(LaunchConfigurationsDialog.java:621)
	at org.eclipse.jface.viewers.StructuredViewer$3.run(StructuredViewer.java:872)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:42)
	at org.eclipse.ui.internal.JFaceUtil.lambda$0(JFaceUtil.java:44)
	at org.eclipse.jface.util.SafeRunnable.run(SafeRunnable.java:173)
	at org.eclipse.jface.viewers.StructuredViewer.firePostSelectionChanged(StructuredViewer.java:869)
	at org.eclipse.jface.viewers.StructuredViewer.setSelection(StructuredViewer.java:1718)
	at org.eclipse.jface.viewers.TreeViewer.setSelection(TreeViewer.java:1075)
	at org.eclipse.jface.viewers.Viewer.setSelection(Viewer.java:380)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog.doInitialTreeSelection(LaunchConfigurationsDialog.java:661)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog.initializeContent(LaunchConfigurationsDialog.java:1150)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog.createContents(LaunchConfigurationsDialog.java:481)
	at org.eclipse.jface.window.Window.create(Window.java:426)
	at org.eclipse.jface.dialogs.Dialog.create(Dialog.java:1096)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog.create(LaunchConfigurationsDialog.java:380)
	at org.eclipse.jface.window.Window.open(Window.java:783)
	at org.eclipse.debug.internal.ui.launchConfigurations.LaunchConfigurationsDialog.open(LaunchConfigurationsDialog.java:1203)
	at org.eclipse.debug.ui.DebugUITools$3.run(DebugUITools.java:629)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
	at org.eclipse.debug.ui.DebugUITools.openLaunchConfigurationDialogOnGroup(DebugUITools.java:637)
	at org.eclipse.debug.ui.DebugUITools.openLaunchConfigurationDialogOnGroup(DebugUITools.java:570)
[... cut ...]
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
```

ilg-ul, 2017-09-14:
thank you, I'll investigate.

ilg-ul, 2017-09-14:
I ran a test with Oxygen and OpenOCD started properly.

the code does some macro subtitutions from the preferences store. are you sure your preferences are ok?

TM1234, 2017-09-14:
Thanks for the quick reply Liviu.
You mean OpenOCD path and executable name string substitution variables?
They look fine and were not changed from defaults.
Let me double check everything just in case I overlooked something.

TM1234, 2017-09-14:
OK - I think I see it now.
Apologies but I DO change the openocd path setting from my plugin_customization.ini referenced by my eclipse.ini.
I set it as follows:

`ilg.gnumcueclipse.debug.gdbjtag.openocd/install.folder=${eclipse_home}/../openocd/bin`

And I see this default override appearing in the GUI under Window > Preferences > MCU > Global OpenOCD Path.

But it looks like using an eclipse variable (i.e. "{eclipse_home}") here causes problems even though this worked in previous versions of the GNU ARM Plugins.

I change the setting to c:\Microsemi\SoftConsole_v5.2\openocd\bin and then I no longer get the error.

Has the code changed in this support since the move from GNU ARM Eclipse to GNU MCU Eclipse?
If eclipse variable expansion is no longer working here at the moment would it be a big deal to (re)instate it?

Thanks a lot.

TM1234, 2017-09-14:
FWIW eclipse variable expansion has always worked and still works in the other fields - e.g. the toolchain paths so the openocd path seems to be the exception.

ilg-ul, 2017-09-14:
> Has the code changed in this support since the move from GNU ARM Eclipse to GNU MCU Eclipse?

at a certain point the macros were reimplemented as dynamic variables, probably at this moment the problem was introduced.

the code is in ilg.gnumcueclipse.debug.gdbjtag.openocd.DynamicVariableResolver, if you can suggest changes to make it work with your configuration, please let me know.

TM1234, 2017-09-15:
Quick update. I get the problem with GNU MCU C/C++ OpenOCD Debugging 4.1.5.201706030906. I can't remember where I got this from but maybe it was some intermediate/work in progress build?

I just installed the latest plugins available from the update site http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/ including GNU MCU C/C++ OpenOCD Debugging 4.1.5.201707111115 and I DO NOT get the problem there.

So apologies for the false alarm and I'll close this now.

closed 2017-09-15 by TM1234