eclipse-embed-cdt/eclipse-plugins #236: Question about openocd path/exe configuration

TM1234, 2017-09-15:
Quick question about the openocd path and exe configuration.

Are these still required and must they be correctly configured:

> Window > Preferences > Run/Debug > String Substitution > openocd_executable | openocd_path

now that the openocd path and exe are configured under

> Window > Preferences > MCU > Global | Workspace OpenOCD Path > Executable | Folder

If so what is the rationale for this seeming duplication?
Is it for backward compatibility?
If backward compatibility is not required can 

> Window > Preferences > Run/Debug > String Substitution > openocd_executable | openocd_path

be left unconfigured?

Thanks.

ilg-ul, 2017-09-15:
if I remember right, there should be no more variables in the String Substitution page, they are now implemented as dynamic variables.

try a fresh Eclipse and a fresh workspace.

TM1234, 2017-09-15:
You are correct.
The string substitutions are no longer required or appear by default.
I was picking them up from some previous workspace or eclipse settings.
Thanks.

closed 2017-09-15 by TM1234

asier70, 2017-10-29:
I have a problem that is appropriate for this thread.
I use OpenOCD compiled by myself for some reason (at windows). I filled Global OpenOCD Path properly. The problem is with scripts that can't be found. I use additional command
-s "${openocd_path}/../scripts"
but ${openocd_path} in it is resolved as empty string.
This trick worked on GNU ARM plugin (with String Subtitutions), but now in GNU MCU plugin with dynamic MCU variables doesn't work. Now only the full path to scripts works.
Any suggestion how to solve this problem?

ilg-ul, 2017-10-29:
it looks macro substitution needs to be done explicitly in a different way. 

I opened https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/246 to keep track of it.