eclipse-embed-cdt/eclipse-plugins #237: Question about update site contents

TM1234, 2017-09-15:
When I pulled the contents of the update site down locally:

` wget -e robots=off --recursive --no-parent https://dl.bintray.com/gnu-mcu-eclipse/v4-neon-updates/`

I notice that there are two copies if every plugin and feature jar with the same version number but different date/time stamps.

```
$ ls -1 features plugins
features:
ilg.gnumcueclipse.codered.feature_1.1.2.201705291517.jar
ilg.gnumcueclipse.codered.feature_1.1.2.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.jlink.feature_4.1.5.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.jlink.feature_4.1.5.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.openocd.feature_4.1.5.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.openocd.feature_4.1.5.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.pyocd.feature_1.1.4.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.pyocd.feature_1.1.4.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.qemu.feature_3.1.5.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.qemu.feature_3.1.5.201707111115.jar
ilg.gnumcueclipse.doc.user.feature_1.1.2.201705291517.jar
ilg.gnumcueclipse.doc.user.feature_1.1.2.201707111115.jar
ilg.gnumcueclipse.managedbuild.cross.arm.feature_2.5.1.201705291517.jar
ilg.gnumcueclipse.managedbuild.cross.arm.feature_2.5.2.201707111115.jar
ilg.gnumcueclipse.managedbuild.cross.riscv.feature_2.5.1.201705291517.jar
ilg.gnumcueclipse.managedbuild.cross.riscv.feature_2.5.2.201707111115.jar
ilg.gnumcueclipse.packs.feature_2.2.2.201705291517.jar
ilg.gnumcueclipse.packs.feature_2.2.2.201707111115.jar
ilg.gnumcueclipse.templates.ad.feature_1.1.5.201705291517.jar
ilg.gnumcueclipse.templates.ad.feature_1.1.5.201707111115.jar
ilg.gnumcueclipse.templates.cortexm.feature_1.4.3.201705291517.jar
ilg.gnumcueclipse.templates.cortexm.feature_1.4.3.201707111115.jar
ilg.gnumcueclipse.templates.freescale.feature_2.2.9.201705291517.jar
ilg.gnumcueclipse.templates.freescale.feature_2.2.9.201707111115.jar
ilg.gnumcueclipse.templates.stm.feature_2.6.3.201705291517.jar
ilg.gnumcueclipse.templates.stm.feature_2.6.3.201707111115.jar
index.html

plugins:
ilg.gnumcueclipse.codered_1.1.2.201705291517.jar
ilg.gnumcueclipse.codered_1.1.2.201707111115.jar
ilg.gnumcueclipse.core_4.1.1.201705291517.jar
ilg.gnumcueclipse.core_4.1.1.201707111115.jar
ilg.gnumcueclipse.debug.core_1.2.2.201705291517.jar
ilg.gnumcueclipse.debug.core_1.2.2.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.jlink_4.1.5.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.jlink_4.1.5.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.openocd_4.1.5.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.openocd_4.1.5.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.pyocd_1.1.4.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.pyocd_1.1.4.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.qemu_3.1.5.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.qemu_3.1.5.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag.restart_1.3.2.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag.restart_1.3.2.201707111115.jar
ilg.gnumcueclipse.debug.gdbjtag_3.2.2.201705291517.jar
ilg.gnumcueclipse.debug.gdbjtag_3.2.2.201707111115.jar
ilg.gnumcueclipse.debug.packs_1.1.2.201705291517.jar
ilg.gnumcueclipse.debug.packs_1.1.2.201707111115.jar
ilg.gnumcueclipse.doc.user_1.1.2.201705291517.jar
ilg.gnumcueclipse.doc.user_1.1.2.201707111115.jar
ilg.gnumcueclipse.managedbuild.cross.arm_2.5.1.201705291517.jar
ilg.gnumcueclipse.managedbuild.cross.arm_2.5.2.201707111115.jar
ilg.gnumcueclipse.managedbuild.cross.riscv_2.5.1.201705291517.jar
ilg.gnumcueclipse.managedbuild.cross.riscv_2.5.2.201707111115.jar
ilg.gnumcueclipse.managedbuild.cross_1.1.1.201705291517.jar
ilg.gnumcueclipse.managedbuild.cross_1.1.2.201707111115.jar
ilg.gnumcueclipse.managedbuild.packs_1.3.2.201705291517.jar
ilg.gnumcueclipse.managedbuild.packs_1.3.2.201707111115.jar
ilg.gnumcueclipse.packs.core_1.1.4.201705291517.jar
ilg.gnumcueclipse.packs.core_1.1.4.201707111115.jar
ilg.gnumcueclipse.packs.data_1.3.2.201705291517.jar
ilg.gnumcueclipse.packs.data_1.3.2.201707111115.jar
ilg.gnumcueclipse.packs.ui_1.2.5.201705291517.jar
ilg.gnumcueclipse.packs.ui_1.2.5.201707111115.jar
ilg.gnumcueclipse.packs_1.2.5.201705291517.jar
ilg.gnumcueclipse.packs_1.2.5.201707111115.jar
ilg.gnumcueclipse.templates.ad_1.1.5.201705291517.jar
ilg.gnumcueclipse.templates.ad_1.1.5.201707111115.jar
ilg.gnumcueclipse.templates.core_2.5.7.201705291517.jar
ilg.gnumcueclipse.templates.core_2.5.7.201707111115.jar
ilg.gnumcueclipse.templates.cortexm_1.4.3.201705291517.jar
ilg.gnumcueclipse.templates.cortexm_1.4.3.201707111115.jar
ilg.gnumcueclipse.templates.freescale.pe_1.2.2.201705291517.jar
ilg.gnumcueclipse.templates.freescale.pe_1.2.2.201707111115.jar
ilg.gnumcueclipse.templates.freescale_2.2.9.201705291517.jar
ilg.gnumcueclipse.templates.freescale_2.2.9.201707111115.jar
ilg.gnumcueclipse.templates.stm_2.6.3.201705291517.jar
ilg.gnumcueclipse.templates.stm_2.6.3.201707111115.jar
index.html
```

In practice this has no obvious adverse impact as the earlier versions never appear in the Eclispe software installation dialogs.

But I just wondered if it was a small (innocuous) anomaly that there were second copies of each of the jar files?

ilg-ul, 2017-09-15:
This looks to be one of the Bintray oddities, they have an unusual way of storing multiple versions of a project.

TM1234, 2017-09-15:
OK - no worries. I was just curious.
I'll close this one so. Thanks.

closed 2017-09-15 by TM1234