eclipse-embed-cdt/eclipse-plugins #239: When macro is in the toolpath then the path priority becomes reversed.

AntonKrug, 2017-09-22:
### Description

[Description of the bug or feature]

### Steps to Reproduce

1. Install ARM toolchain belong side the eclipse install directory 

2. Create/Touch file under this path: 
C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update/bin/arm-none-eabi-gcc.exe 
To emulate that we installed official ARM toolchain as well.
 
3. Add this path with PluginCustomization to the eclipse and use the macro to reference eclipse's home directory:
ilg.gnumcueclipse.managedbuild.cross.arm/toolchain.path.1287942917=${eclipse_home}/../arm-none-eabi-gcc/bin

4. Right click on ARM project which does not have specific settings overridden from a default (the one which takes the global settings). And either got to MCU -> ARM Toolchains Paths, or to C/C ++ Build -> Settings -> Toolchains -> Toolchain path.

**Expected behaviour:** 
It should have used the ${eclipse_home}/../arm-none-eabi-gcc/bin

**Actual behaviour:** 
But actually it will contain the:
C:\Program Files (x86)\GNU Tools ARM Embedded\6 2017-q2-update\bin

### Versions

* [plug-in version]
  GNU MCU C/C++ ARM Cross Compiler	2.5.2.201709220847	ilg.gnumcueclipse.managedbuild.cross.arm.feature.feature.group	Liviu Ionescu
  GNU MCU C/C++ OpenOCD Debugging	4.2.1.201709220847	ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.feature.group	Liviu Ionescu
* [Eclipse version]
Oxygen
* [Java version]
1.8
* [operating system]
Windows 10
* [toolchain version]
gcc version 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437] (GNU Tools for ARM Embedded Processors 6-2017-q2-update)

### Verbose info

When I modify this file:

/ilg.gnumcueclipse.managedbuild.cross.arm/src/ilg/gnumcueclipse/managedbuild/cross/arm/preferences/DefaultPreferenceInitializer.java

`public void finalizeInitializationsDefaultPreferences() {

	if (Activator.getInstance().isDebugging()) {
		System.out.println("arm.LateInitializer.finalizeInitializationsDefaultPreferences()");
	}

	DefaultPreferences deprecatedDefaultPreferences = new DefaultPreferences(
			"ilg.gnuarmeclipse.managedbuild.cross");

	// Toolchains paths
	for (ToolchainDefinition toolchain : ToolchainDefinition.getList()) {

		String toolchainName = toolchain.getName();

		// If the search path is known, discover toolchain.
		int ix;
		try {
			ix = ToolchainDefinition.findToolchainByName(toolchainName);
		} catch (IndexOutOfBoundsException e) {
			ix = ToolchainDefinition.getDefault();
		}

		System.out.println("------------------ " + ix + " - " + toolchainName + " --------------------------");

		String executableName = ToolchainDefinition.getToolchain(ix).getFullCmdC();
		System.out.println(executableName);

		// Try the defaults from the GNU MCU Eclipse store.
		String path = fDefaultPreferences.getToolchainPath(toolchainName);
		System.out.println("1-default-preference. Setting path to: " + path);

		if (!fDefaultPreferences.checkFolderExecutable(path, executableName)) {
			// Try the deprecated GNU ARM Eclipse store.
			path = deprecatedDefaultPreferences.getToolchainPath(toolchainName);
			System.out.println("2-deprecatedDefault, previous path was not found, overiding it with: " + path);
		}

		if (!fDefaultPreferences.checkFolderExecutable(path, executableName)) {
			// Try the persistent preferences.
			path = fPersistentPreferences.getToolchainPath(toolchainName, null);
			System.out.println("3-persistentPreferences, previous path was not found, overiding it with: " + path);
		}

		if (!fDefaultPreferences.checkFolderExecutable(path, executableName)) {
			// If not defined elsewhere, discover.
			path = fDefaultPreferences.discoverToolchainPath(toolchainName, executableName);
			System.out.println("4-discoverToolchainPath. previous path was not found, overiding it with: " + path);
		}
		
		System.out.println();

		if (path != null && !path.isEmpty()) {
			// If the toolchain path was finally discovered, store
			// it in the default preferences.
			fDefaultPreferences.putToolchainPath(toolchainName, path);
		}

		if (Activator.getInstance().isDebugging()) {
			System.out.println("arm.LateInitializer.finalizeInitializationsDefaultPreferences() done");
		}
	}
}
`

And this file:
/ilg.gnumcueclipse.core/src/ilg/gnumcueclipse/core/preferences/DefaultPreferences.java

`
	public boolean checkFolderExecutable(String folder, String executableName) {

		if (folder == null || folder.isEmpty()) {
			return false;
		}
		if (executableName == null || executableName.isEmpty()) {
			return false;
		}

		if (EclipseUtils.isWindows()) {
			executableName = addExeExtension(executableName);
		}

		IPath path = (new Path(folder)).append(executableName);
		
		System.out.println("Testing if this exists: " + path.toString());
		
		if (path.toFile().isFile()) {
			return true;
		}

		return false;
	}
`

Adding more printouts will show me the logic how it tries to evaluate the path:

> ------------------ 0 - GNU Tools for ARM Embedded Processors --------------------------
> arm-none-eabi-gcc
> 1-default-preference. Setting path to: ${eclipse_home}/../arm-none-eabi-gcc/bin
> Testing if this exists: arm-none-eabi-gcc/bin/arm-none-eabi-gcc.exe
> 2-deprecatedDefault, previous path was not found, overiding it with:
> 3-persistentPreferences, previous path was not found, overiding it with: ${eclipse_home}/../arm-none-eabi-gcc/bin
> Testing if this exists: arm-none-eabi-gcc/bin/arm-none-eabi-gcc.exe
> DefaultPreferences.searchLatestExecutable("${user.home}/AppData/Local/GNU Tools ARM Embedded;${user.home}/opt/GNU Tools ARM Embedded;${user.home}/opt/gnu-tools-arm-embedded;${user.home}/local/GNU Tools ARM Embedded;${user.home}/local/gnu-tools-arm-embedded;C:/opt/GNU Tools ARM Embedded;C:/opt/gnu-tools-arm-embedded;C:/Program Files/GNU Tools ARM Embedded;C:/Program Files (x86)/GNU Tools ARM Embedded;D:/Program Files/GNU Tools ARM Embedded;D:/Program Files (x86)/GNU Tools ARM Embedded", bin", arm-none-eabi-gcc)
> 4-discoverToolchainPath. previous path was not found, overiding it with: C:/Program Files (x86)/GNU Tools ARM Embedded/6 2017-q2-update/bin
> 

In essence, the first preference becomes the very last because with the macro it will not get resolved at all and it will fail every single time, meaning that anything else will be taken rather that the specified path. And if I have the ARM toolchain present (or even a dummy file with that name) it will rather take that then the path we want.

I think the similar situation applies to the RISC-V bundle as well.

---

Note: Remove unnecessary text after reading, before entering your ticket.

# @

ilg-ul, 2017-09-22:
I guess you are right, although I'm not sure I understand the actual problem

do you have a clear suggestion?

AntonKrug, 2017-09-22:
I'm not 100% sure where you are resolving macros, but you might need to do it at the checkFolderExecutable as well. Or some of the other routines might not work as designed. It will not be noticeable until the right conditions are met. Even when it's working right it's actually by accident, not by design.

ilg-ul, 2017-09-22:
in other words, you suggest to resolve macros in the toolchain discovery code? so that the search will end when the local toolchain is identified, and not return the default?

AntonKrug, 2017-09-22:
No, discovery code is already too late. I think the **fDefaultPreferences.checkFolderExecutable** shouldn't fail every time just because the path contains a macro.

ilg-ul, 2017-09-22:
ok, I'll investigate if `checkFolderExecutable()` can also expand macros.

ilg-ul, 2017-09-22:
see https://github.com/gnu-mcu-eclipse/eclipse-plugins/commit/6e745e1858174e700d576ff0d3557a0ac9f65ce5

TM1234, 2017-10-20:
Hi Liviu - is  there a binary build of the plugins with this fix available from an update site or do I need to pull the sources and build a snapshot myself? Thanks.

ilg-ul, 2017-10-20:
yes, I just published the latest version I have, try 

http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

TM1234, 2017-10-20:
Great! Thanks a lot. I'll try that this afternoon.

ilg-ul, 2017-11-10:
fied since v4.2.1-201711101735

closed 2017-11-10 by ilg-ul