eclipse-embed-cdt/eclipse-plugins #242: Leaving out the telnet port empty gives cryptic error.

AntonKrug, 2017-10-18:
### Description

When a required field is left empty it returns a cryptic error message.

### Steps to Reproduce

1. Setup generic project with the OpenOCD luncher.
2. Opening a debug launcher settings and deleting content from:
Debugger -> OpenOCD Setup -> Telnet port 
will cause getting a strange error dialog:

**Expected behaviour:**
Less cryptic error description or not allowing the user to put an empty string on a required field (instead of an InputStringField use NumberTextField, if that would help). 

**Actual behaviour:** 
"Update LCD" has encountered a problem.
An internal error has occurred.
An internal error has occurred.
  For input string: ""

[image]

(googling his error points to workspace/maven issues which are not this case, so it can create even more confusion)

### Versions
* GNU MCU C/C++ OpenOCD Debugging 4.2.1
* Eclipse 4.6
* Java 1.8
* Windows 10 - 64bit

ilg-ul, 2017-10-18:
I'll take a look.

AntonKrug, 2017-10-18:
I was trying to find the text "Update LCD" in the repo and I couldn't find it, so my guess is that your method triggers exception which gets caught outside your plugin by something else.

ilg-ul, 2017-10-18:
I don't remember this message either, so I guess you are right, the error comes from outside the plugin.

anyway, you are right, the plugin should validate the content of the gui input.

AntonKrug, 2017-10-18:
I could be wrong but my guess would be that something like this causes an exception:

Integer.parseInt(“”);

Which is not caught by the plugin and gets propagated outside.

ilg-ul, 2017-10-18:
yes, it might be so.

I don't have time to investigate now, so feel free to suggest a PR.

ilg-ul, 2017-11-10:
fixed since v4.2.1-201711101735

closed 2017-11-10 by ilg-ul

AntonKrug, 2017-11-13:
Excellent work :-)