eclipse-embed-cdt/eclipse-plugins #244: mbed CLI exports and standard includes

chrylis, 2017-10-23:
I'm running Arch Linux on amd64. I have an mbed project I can build from the command line, and I've installed the Eclipse plugin per the docs, including setting the toolchain path.

However, when I export my mbed project and import into my Eclipse workspace, I don't have an include path for the standard includes. I'm not familiar enough with CDT to know whether this is because the mbed CLI should be including entries in the `.cproject` corresponding to the `GCC_ARM_HOME` or because CDT should be importing "standard" includes from another provider. In either case, it looks like there's a setup step that's missing. Could you let me know whether I should check one of these boxes, file a bug report on mbed to add standard include paths to the generated `.cproject`, or do something else?

[image]

ilg-ul, 2017-10-23:
I don't know, probably you have a custom Eclipse configuration, by custom meaning it was not installed according to the GNU MCU Eclipse manual.

The standard include folders are shown only when the plugin is able to start the compiler, so I would check the toolchain path.

To know which provider you need to select, create a blinky project using, for example the F4 template, and inspect the result.

chrylis, 2017-10-23:
> by custom meaning it was not installed according to the GNU MCU Eclipse manual

This is a clean CDT installation (Oxygen.1), and I followed the installation instructions exactly.

I created a Hello World, and it seems that the "CDT ARM Cross GCC Built-in Compiler Settings" provider was checked. I am guessing that the mbed project generator needs to include an entry for it.

ilg-ul, 2017-10-23:
> I followed the installation instructions exactly.

did you get a functional blinky project running on qemu, as per the blinky tutorial?

> the "CDT ARM Cross GCC Built-in Compiler Settings" provider 

if you manually check it, do you see the include folders?

chrylis, 2017-10-23:
> did you get a functional blinky project running on qemu, as per the blinky tutorial?

I did not run through the qemu tutorial because I'm using mbed instead of the CMSIS Pack system. I exported/imported a basic mbed blinky program and got it working on a development board and only ran into trouble when I tried to import `<stdint.h>` for `uint8_t`.

> if you manually check it, do you see the include folders?

Yes, the folders show up, but I also get hundreds of thousands of errors on the mbed codebase. (At least a lot of them seem to be justified, but seeing that is why I wasn't sure that checking the box was the correct solution.)

It looks like this is just an update that needs to go into the project exporter. Thanks for your help!

closed 2017-10-23 by chrylis

ilg-ul, 2017-10-23:
could you do again an export, save the `.project` and `.cproject` files, then edit the configuration to add the provider and finally compare the two files, to see the differences?