eclipse-embed-cdt/eclipse-plugins #246: Substitute ${openocd_path} in openocd gui fields

ilg-ul, 2017-10-29:
https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/236#issuecomment-340304450

ilg-ul, 2017-11-10:
fixed since v4.2.1-201711101735

closed 2017-11-10 by ilg-ul

asier70, 2017-11-13:
Confirmed - it works. Thanks!

ilg-ul, 2017-11-13:
great, thank you.

biswaMW, 2018-05-03:
Is this by chance broken in v4.3.2.201804191501 ?
I see : Cannot run program "${openocd_path}/${openocd_executable}": Launching failed

Both the Global and the Workspace paths for OpenOCD are set correctly.
Using those paths directly in the Debug window works though.

Any ideas ?
Running neon.3 with GNU MCU Eclipse 4.3.3

ilg-ul, 2018-05-03:
I don't remember having this problem in 4.3.2.

Please download the latest version and try again in a clean Eclipse and a clean workspace.

TM1234, 2018-05-03:
This is not broken for me.
I presume that in the debug launch configuration GUI on the Debugger tab you see Executable path: ${openocd_path}/${openocd_executable}?
What do you see in the Actual executable field and is it correct?

biswaMW, 2018-05-03:
How do I examine the 'Actual executable field' ?

From the error window this is what I have.
Launching command [${openocd_path}/${openocd_executable} -c gdb_port 3333 -c telnet_port 4444 -c tcl_port 6666 -f board/st_nucleo_f3.cfg -c echo "Started by GNU MCU Eclipse"] failed.

Are the macros resolved internally (i.e. not visible in the error window) ?

biswaMW, 2018-05-03:
@ilg-ul : I'm sorry I missed the 'Actual executable' field. I do see it. (I was just used to the old plug-in. I recently upgraded my machine and was installing the latest plug-in.).

It contains : ${openocd_path}/${openocd_executable}

If it helps, I'm running Win10, JDK1.8 and already tried a clean re-install with a new workspace.

Thanks for looking into this !

ilg-ul, 2018-05-03:
you somehow messed your Eclipse and this affect the macro substitutions.

go to both Global/Workspace OpenOCD paths and Restore Defaults.

if this does not help, install in a separate folder the latest Eclipse which includes the latest plug-ins, do not install any other plug-ins, and retry.

TM1234, 2018-05-03:
You should see something like this (the specific paths will be different so just for illustration):
[image]
[image]

biswaMW, 2018-05-03:
@TM1234 @ilg-ul  : Thanks for the suggestions !
After some trial and error, it looks like it isn't an issue with install at all.
It might have to do with my workflow.

I do not use Eclipse to build/develop applications (so I don't create/import projects)
I just create a standalone debug configuration, load an elf, specify the board to OpenOCD and start debugging.
[image]

-------------------------------------------------------
This time I tried creating (just as an experiment) a new project. Now when I generate a debug configuration associated with this project, the 'Actual Executable' field does indeed populate appropriately and everything works just fine !
[image]

I am familiar with the approach to 'import' an elf and create a (non-buildable) project to debug the exec.
So I tried this and it worked as well.

In essence, it looks like a debug config now needs to be associated with a project. 
Is this expected ?
Should I switch to *always* creating projects with the new plugin ?

(I speculate my original workflow was working with the former plug-in since the macros were string substituted, which isn't the case now.)

Boy, was it a long session today and thanks a ton for helping !

ilg-ul, 2018-05-03:
> it looks like a debug config now needs to be associated with a project.
> Is this expected ?

well, a debug config does not **need** to be associated with a project.

you can always browse for the absolute address of the binaries.

however, the defaults used to create the debug config expect a project, since expanding the macos require a project.

but, as I said, you can always override the devaults with any setting you want.