eclipse-embed-cdt/eclipse-plugins #248: Check and fix the J-Link GDB Server termination logic

ilg-ul, 2017-11-05:
The termination logic is complicated, since it needs to terminate all processes regardless which one was killed or terminated.

There may be a bug, that kills the GDB server too early, prevnting its termination logic to run, to leave the device running.

TM1234, 2018-01-25:
Hi Liviu - any chance this is a bug in Eclipse's starter.exe e.g. ...\eclipse\plugins\org.eclipse.cdt.core.win32.x86_64_5.4.1.201712020452\os\win32\x86_64\starter.exe?

We had to modify this to terminate openocd more gracefully than it would normally - see the attached modified version paying attention to the bits tagged "Microsemi".

I need to submit this (or a more general fix) back to the project for consideration as I consider starter.exe broken as it stands.

If this is not relevant to the original issue then apologies for the distraction! 

[eclipse_cdt_starter.zip](https://github.com/gnu-mcu-eclipse/eclipse-plugins/files/1665570/eclipse_cdt_starter.zip)

ilg-ul, 2018-01-25:
I don't know, I thought the original problem is specific to my debug plug-ins. 

There are many internal threads doing various small tasks related to the debugger, and the logic that keeps everything together is very complicated.

TM1234, 2018-01-25:
OK - on reflection, starter.exe is specific to Windows (neither it nor any equivalent exists in CDT for Linux), so if you problem is not specific to Windows then starter.exe is not the problem here even if the way that it tries to terminate launched apps (e.g. external tools etc.) is flawed.

ilg-ul, 2018-01-26:
Yes, it makes sense.