eclipse-embed-cdt/eclipse-plugins #249: GDB OpenOCD Debugging plugin - support configuration of OpenOCD tcl port?

TM1234, 2017-11-09:
Possible small enhancement... admittedly for a possibly rare/edge case scenario.

The GDB OpenOCD Debugging plugin allows the OpenOCD GDB (Remote Serial Protocol) and Telnet ports to be configured (defaulting to 3333 and 4444 respectively). 
OpenOCD also creates/uses a Tcl port which defaults to 6666.
In (possibly rare?) cases where somebody wants to run simultaneous OpenOCD sessions using separate JTAG and target hardware they need to configure the "second" session to use different GDB, Telnet and Tcl ports (e.g. 3334, 4445 and 6667).
Since there is no way to configure the Tcl port in the GUI a second OpenOCD debug session will probably fail unless --command "tcl_port 6667" is passed to OpenOCD - e.g.

--command "tcl_port 6667" --file board/myboard.cfg

ilg-ul, 2017-11-09:
do you suggest to add a new GUI element to configure the Tcl port?

TM1234, 2017-11-09:
Sorry - I should have been clearer - yes - that's what I meant - the addition of a new field to allow the Tcl port (default 6666) to be configured/overridden like the GDB and Telnet ports.

ilg-ul, 2017-11-09:
ok, I'll see if I can add it to the next release.

TM1234, 2017-11-09:
Thanks Liviu - definitely not critical - and I doubt that too many people will use multiple simultaneous openocd debug sessions? - but it might be handy in the rare cases where this is required.

ilg-ul, 2017-11-10:
please try 4.2.1-201711101401 from:

http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

and let me know if it is ok.

ilg-ul, 2017-11-10:
fixed since v4.2.1-201711101735

closed 2017-11-10 by ilg-ul

TM1234, 2017-11-28:
Very minor issue but "TCL port" should probably be "Tcl port"?
Also the tooltip says that the service can be disabled - does that mean by literally entering "disabled" (without quotes) into the input field?
Thanks.

TM1234, 2017-11-28:
> does that mean by literally entering "disabled" (without quotes) into the input field?

Sorry - RTFM - I see now that this is the case:

http://openocd.org/doc/html/Server-Configuration.html

TM1234, 2017-11-28:
Also - should the Telnet port field also allow "disabled" to be entered?
At the moment it doesn't.
For OpenOCD JTAG debugging I think that the only mandatory port is the GDB [Remote Serial Protocol] port?

ilg-ul, 2017-11-28:
> I can accept it there, but is openocd happy with this?

TM1234, 2017-11-28:
Yes - I ran a debug session with Tcl port = disabled in the GUI and Telnet port = 4444 but also with -c "telnet_port disabled" in my openocd command line options and the latter seems to have overridden the GUI setting because when I tried to telnet into openocd I got this:

```
> telnet localhost 4444
Connecting To localhost...Could not open connection to the host, on port 4444: Connect failed
```

In contrast when I leave the GUI default as Telnet port = 4444 then I get this:

```
> telnet localhost 4444
Connecting To localhost...

Open On-Chip Debugger
> reg pc
pc (/32): 0x800016E4
>
```

BTW - having the ability to telnet into openocd while debugging with GDB is sometimes useful although obviously a way to screw things up too! :-)

TommyMurphyTM1234, 2020-08-28:
Hi @ilg-ul 

Is this issues tracker still valid for the GME plugins?
If so a small issue that I noticed relevant to this ticket.
In the GME OpenOCD debugging plugin I can enter "disabled" for the Tcl port and it accepts it but I cannot enter "disabled" for the Telnet port.
When I try to enter anything other than decimal digits I get an error popup.

Regards
Tommy

ilg-ul, 2020-08-28:
What exactly is your proposal?

TommyMurphyTM1234, 2020-08-28:
Allow "disabled" in all port fields same as is allowed in the Tcl field right now?
I suspect that the change is probably trivial but we can look at it and propose a PR if you want?

ilg-ul, 2020-08-29:
> propose a PR

sure. if it comes in the next weeks, it'll probably be included in 5.1.2, planned to be released with 2020-09.