eclipse-embed-cdt/eclipse-plugins #252: Super minor typo: commercial, not comercial

amosnier, 2017-11-27:
At https://gnu-mcu-eclipse.github.io/arch/riscv/arty-reflash/, replace:

"huge, Windows-only comercial application"

by:
"huge, Windows-only commercial application"

Apart from that thanks for the warning, that pretty much eliminates the FPGA option for me. I will have to wait a month for my HiFive1. In the mean time, I will try to write some code in GNU MCU Eclipse. And thanks for the name change (GNU MCU Eclipse): that's what made me realize the existence of RISC-V.

Best regards,

Alain Mosnier

ilg-ul, 2017-11-27:
thank you for spotting the typo, I fixed it.

and I'm glad my site helped you discover RISC-V, it is an interesting architecture.

until you receive the board, you can take a look at the projects generated by the SiFive1 template.

ilg-ul, 2017-11-27:
if you discover more typos, please use the tracker associated with the web project.

closed 2017-11-27 by ilg-ul

amosnier, 2017-11-27:
OK, sorry.