eclipse-embed-cdt/eclipse-plugins #255: MCU Build Tools Path appears in breakpoint properties

flit, 2017-12-13:
### Description

MCU Build Tools Path preferences page appears in breakpoint and watchpoint properties windows.

### Steps to Reproduce

1. Create a breakpoint or watchpoint.
2. Open the breakpoint properties window. You can also open this window as you create the breakpoint/watchpoint by executing one of the Add {Line,Function,Event} Breakpoint… or Add Watchpoint… menu items.

**Expected behaviour:**

MCU->Build Tools Path does not appear in the properties page list.

**Actual behaviour:**

MCU->Build Tools Path does, in fact, appear in the properties page list.

### Versions

* 3b8ab40e84
* Oxygen.1a (4.7.1a)
* 1.8.0_05-b13
* macOS 10.11.6

ilg-ul, 2017-12-13:
could you suggest a PR?

flit, 2017-12-13:
Well, I'd be happy to, but I don't know anything about how the property pages are matched and filtered in Eclipse.

ilg-ul, 2017-12-13:
As many other things in Eclipse, there are various ways to do it :-(

Sometimes it works by adding some conditions in the plugin.xml, sometimes only Java code can do it.

Time permitting, I'll investigate.

flit, 2017-12-13:
Understood, just wanted to make sure it was documented.

ilg-ul, 2018-01-05:
this is a specific case of #262.

closed 2018-01-05 by ilg-ul