eclipse-embed-cdt/eclipse-plugins #257: gnu-mcu-eclipse-che

loverdeg-ep, 2017-12-21:
### Description

What needs to happen for gnu-mcu-eclispe-che, or alternatively clang-mcu-eclispe-che to exist

Thoughts:
* From what I've gathered writing a lanuguage server for GCC would be burdensome
* clang has [clangd ](https://clang.llvm.org/extra/clangd.html) or a supposedly more complete language server implementation [cquery](https://github.com/jacobdufault/cquery)
* Both the cloud IDEs I've evaluatied (Che and C9.io) are naturally focused on "non-cross-compile" and "non-remote debug scenarios"
* I have no understanding of debug for Che currently other than the ideal solution would be a GDB server hooked to the target board/boards.

ilg-ul, 2017-12-21:
I've heard about Eclipse Che, but had no tme to evaluate it.

I'm not even sure that this configuration is useful for embedded systems development, especially on debugging issues.

If you have more time to investigate, please let me know.

loverdeg-ep, 2017-12-21:
Take a quick scan of [Docs Intro](https://www.eclipse.org/che/docs/) if you have the time.
The IDE doesn't need to be browser based Che can serve workspaces to standard eclipse as well.

[image]

I think it would be extremely useful for embedded development considering building an embedded dev environment is so tedious and learning how to build an embedded dev environment is even more tedious. (Though lately this has become somewhat easier due to your efforts in particular)

When working as a team on embedded projects I'd much prefer setting up my workspace on a server somewhere and then just handing my peers a single link which will allow them to start developing immediately. 
This is versus directing them to the gnu-mcu-eclipse documentation and hoping they make it through okay or having giving them a copy of my dev virtual machine which goes immediately out of sync with my copy as soon as they start updating packages and install their preferred tools.

Also, If you haven't been monitoring the [Language Server Protocol](https://www.eclipse.org/community/eclipse_newsletter/2017/april/article5.php) i'd take a look at that too.