eclipse-embed-cdt/eclipse-plugins #258: Eclipse hangs when using variables in GDB start commands

flit, 2017-12-24:
### Description

If one uses variables in the GDB start commands in a launch config, Eclipse will deadlock between the GNU MCU Eclipse thread and the main thread when trying to resolve the variables.

Stack traces of the deadlocked threads:

**Thread [org.eclipse.cdt.dsf.gdb - 105] (Suspended):**
```
waiting for: RunnableLock  (id=163835)	
Object.wait(long) line: not available [native method]	
RunnableLock(Object).wait() line: 502	
UISynchronizer(Synchronizer).syncExec(Runnable) line: 250	
UISynchronizer.syncExec(Runnable) line: 144	
Display.syncExec(Runnable) line: 4870	
SelectedResourceManager.getSelectedResource() line: 129	
SelectedResourceResolver.resolveValue(IDynamicVariable, String) line: 33	
DynamicVariable.getValue(String) line: 56	
StringSubstitutionEngine.resolve(StringSubstitutionEngine$VariableReference, boolean, boolean, IStringVariableManager) line: 269	
StringSubstitutionEngine.substitute(String, boolean, boolean, IStringVariableManager) line: 192	
StringSubstitutionEngine.performStringSubstitution(String, boolean, boolean, IStringVariableManager) line: 87	
StringVariableManager.performStringSubstitution(String, boolean) line: 591	
StringVariableManager.performStringSubstitution(String) line: 357	
ProjectResolver(ResourceResolver).getSelectedResource(IDynamicVariable) line: 128	
ProjectResolver(ResourceResolver).resolveValue(IDynamicVariable, String) line: 48	
DynamicVariable.getValue(String) line: 56	
StringSubstitutionEngine.resolve(StringSubstitutionEngine$VariableReference, boolean, boolean, IStringVariableManager) line: 269	
StringSubstitutionEngine.substitute(String, boolean, boolean, IStringVariableManager) line: 192	
StringSubstitutionEngine.performStringSubstitution(String, boolean, boolean, IStringVariableManager) line: 87	
StringVariableManager.performStringSubstitution(String, boolean) line: 591	
DebugUtils.resolveAll(String, Map<String,Object>) line: 249	
DebuggerCommands.addStartRestartCommands(boolean, List<String>) line: 176	
DebuggerCommands.addGnuMcuStartCommands(List<String>) line: 97	
GnuMcuFinalLaunchSequence_7_2(GnuMcuFinalLaunchSequence).stepGnuMcuStart(RequestMonitor) line: 374	
NativeMethodAccessorImpl.invoke0(Method, Object, Object[]) line: not available [native method]	
NativeMethodAccessorImpl.invoke(Object, Object[]) line: 62	
DelegatingMethodAccessorImpl.invoke(Object, Object[]) line: 43	
Method.invoke(Object, Object...) line: 483	
ReflectionSequence$ReflectionStep.execute(RequestMonitor) line: 155	
GnuMcuFinalLaunchSequence_7_2(Sequence).executeStep(int) line: 459	
Sequence.access$2(Sequence, int) line: 373	
Sequence$2.handleSuccess() line: 420	
Sequence$2(RequestMonitor).handleCompleted() line: 385	
RequestMonitor$2.run() line: 312	
DefaultDsfExecutor$TracingWrapperRunnable.run() line: 374	
Executors$RunnableAdapter<T>.call() line: 511	
ScheduledThreadPoolExecutor$ScheduledFutureTask<V>(FutureTask<V>).run() line: 266	
ScheduledThreadPoolExecutor$ScheduledFutureTask<V>.access$201(ScheduledThreadPoolExecutor$ScheduledFutureTask) line: 180	
ScheduledThreadPoolExecutor$ScheduledFutureTask<V>.run() line: 293	
DefaultDsfExecutor(ThreadPoolExecutor).runWorker(ThreadPoolExecutor$Worker) line: 1142	
ThreadPoolExecutor$Worker.run() line: 617	
Thread.run() line: 745	
…
```

**Thread [main] (Suspended):**
```
owns: RunnableLock  (id=160268)	
waiting for: Query$QueryRm  (id=160269)	
Object.wait(long) line: not available [native method]	
Query$QueryRm(Object).wait() line: 502 [local variables unavailable]	
GdbSteppingModeTarget$1(Query<V>).get() line: 102	
GdbSteppingModeTarget.supportsInstructionStepping() line: 63	
ToggleInstructionStepModeActionDelegate.debugContextChanged(DebugContextEvent) line: 149	
DebugWindowContextService$2.run() line: 231	
SafeRunner.run(ISafeRunnable) line: 42	
DebugWindowContextService.notify(DebugContextEvent, ListenerList<IDebugContextListener>) line: 228	
DebugWindowContextService.notify(DebugContextEvent) line: 204	
DebugWindowContextService.debugContextChanged(DebugContextEvent) line: 452	
AbstractDebugContextProvider$1.run() line: 82	
SafeRunner.run(ISafeRunnable) line: 42	
LaunchView$ContextProviderProxy(AbstractDebugContextProvider).fire(DebugContextEvent) line: 79	
LaunchView$ContextProviderProxy.debugContextChanged(DebugContextEvent) line: 519	
AbstractDebugContextProvider$1.run() line: 82	
…
```

### Steps to Reproduce

1. Create a debug launch config (pyOCD in my case).
2. Add a variable to the GDB start commands, such as `${project_loc}`.
3. Save the launch config and launch it.

**Expected behaviour:**

Launch config launches successfully and GDB executes the start commands with resolved variables.

**Actual behaviour:**

Eclipse hangs with the above described deadlock.

### Versions

- Plug-in version: 3b8ab40
- Eclipse version: Oxygen.1a (4.7.1a)
- Java version: 1.8.0_05-b13
- OS version: macOS 10.11.6

ilg-ul, 2017-12-24:
this sounds bad, I currently have no idea what to do... :-(

flit, 2017-12-24:
Could `DebugUtils.resolveAll()` put a runnable on the main thread to resolve the variables and wait for it to complete?

ilg-ul, 2017-12-24:
if you find a working solution, please post it as a PR.

IIXII78, 2019-04-01:
Found this report here while searching for duplicates for a problem I just reported, see https://bugs.eclipse.org/bugs/show_bug.cgi?id=546000

ilg-ul, 2019-04-01:
It looks like an Eclipse CDT problem; so not much to do here.