eclipse-embed-cdt/eclipse-plugins #269: assert_param() failed: file "../system/src/stm32f7-hal/stm32f7xx_hal_rcc_ex.c", line 1256

jiapei100, 2018-01-19:
Hi, all:

I'm testing a STM32 NUCLEO F767ZI board through Ubuntu Linux host. And, I'm using Eclipse + gnu-mcu-eclipse . Actually, my code is just the default blinky project, so, it's NOT hard to create the project and successfully build it.

However, whenever I tried to debug/run it, I always met the following error message:

```
SEGGER J-Link GDB Server V6.22d - Terminal output channel
assert_param() failed: file "../system/src/stm32f7-hal/stm32f7xx_hal_rcc_ex.c", line 1256
abort(), exiting...
Connection closed by the GDB server.
```

**line 1256** of file **stm32f7xx_hal_rcc_ex.c** is:
```
assert_param(IS_RCC_PLLR_VALUE(RCC_OscInitStruct->PLL.PLLR));
```

In the corresponding header file **stm32f7xx_hal_rcc_ex.h**, there is a definition about **PLL>PLLR**: 
```
#if defined (STM32F765xx) || defined (STM32F767xx) || defined (STM32F769xx) || defined (STM32F777xx) || defined (STM32F779xx)
  uint32_t PLLR;       /*!< PLLR: Division factor for DSI clock.
                            This parameter must be a number between Min_Data = 2 and Max_Data = 7    */
#endif /* STM32F767xx || STM32F769xx || STM32F777xx || STM32F779xx */  
```

I then tried to add
**#define STM32F767xx 1**
in front of everything in **main.c**, I still get this error message.

For me, it's just because the Macro **STM32F767xx** has **NEVER** been defined.

Can anybody give me a hand ?

Thank you very much

Pei

ilg-ul, 2018-01-19:
If you think that this is related to a bug in GNU MCU Eclipse STM32F7 template, please provide all details to reproduce it.

However, please note that I might not be able to actually run the code generated by the template.

Also please note that only the F4 Blinky projects are guaranteed to run, since I test them on QEMU and the physical board, the other... more or less...

ilg-ul, 2018-01-19:
(After reading your message twice) I think I understand that you ask why the `STM32F767xx` is not defined.

Did you check the Setting -> Symbols page? What symbol is defined there?

jiapei100, 2018-01-20:
@ilg-ul 

Thank you so much for your prompt reply... I do have the symbol **STM32F767xx** defined in Project **Properties**~**Settings**~**Cross ARM GNU C Compiler**~**Preprocessor**.

I NOW think your **FIRST** reply might be the correct answer: F4 Blinky is guaranteed to run, but F7 Blinky is **NOT** ??? There might be a bug in **GNU MCU Ecipse STM32F7** template ??

What I've done is just 
**File**~**New**~**C/C++ Project**~**C Managed Build**~**Executable STM32F7xx C/C++ Project  + CROSS ARM GCC** (Type in Project Name: F767ZI_1 )~**Chip family STM32F767xx, Flash Size(kB) 2048, External clock (Hz) 8000000**, Then, **Next**~**Next**~**Next**...

Finally, select **Cross GNU ARM Toolchina**, which is directly downloaded from [ARM Official Website](https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads), the **NEWEST** version
**gcc-arm-none-eabi-7-2017-q4-major Linux 64-bit** .

Afterwards, **Finish**.

After that, right click **F767ZI_1**~**Run AS/Debug As**~**Run Configuration/Debug Configuration** ~**GDB SEGGER J-Link Debugging**~**New** (Automatically generate a Name **F767ZI_1 Debug**) ~in Tab **Debugger** make sure the Device name **STM32F767ZI**. 

Finally, click **Run** or **Debug**.

That's how I generate the code... **Build Project** successfully. But, **Run** or **Debug** generate this **ERROR** message:

> SEGGER J-Link GDB Server V6.22d - Terminal output channel
> assert_param() failed: file "../system/src/stm32f7-hal/stm32f7xx_hal_rcc_ex.c", line 1256
> abort(), exiting...

Looking forward to your suggestions...

Cheers

ilg-ul, 2018-01-20:
You are right, the F7 template has a bug, the PLLR field is not initialised.

I fixed it and published 4.3.2-201801200943, please test it with "Install New Software..." from 

http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

For existing projects, add the following in `_initialize_hardware.c` after line 179:

```
#if defined (STM32F765xx) || defined (STM32F767xx) || defined (STM32F769xx) || defined (STM32F777xx) || defined (STM32F779xx)
  /* PLLR: Division factor for DSI clock.
     This parameter must be a number between Min_Data = 2 and Max_Data = 7    */
  RCC_OscInitStruct.PLL.PLLR = 2;
#endif /* STM32F767xx || STM32F769xx || STM32F777xx || STM32F779xx */
```

Possibly adjust the value for your application.

jiapei100, 2018-01-20:
@ilg-ul 

That's great ilg-uI. Thank you vey much for your prompt patch...
The code now reports **NO** bug, but with the following output:

F767ZI_Blinky Debug [GDB SEGGER J-Link Debugging] Semihosting and SWV
```
SEGGER J-Link GDB Server V6.22d - Terminal output channel
8
```

However, I didn't see any LED on NUCLEO STM32F767ZI blinking... But, the LED on SEGGER J-Link seems to be blinking... Is that the expected result? (Sorry for my naive question.)

BTW, is there any camera exsample (for example, ov5640) for me to follow?

Cheers
Pei

jiapei100, 2018-01-20:
@ilg-ul 

BTW, after I stopped **running/debuging**, I got the following info, can you please have a look at it? Is that correct?

```
SEGGER J-Link GDB Server V6.22d Command Line Version

JLinkARM.dll V6.22d (DLL compiled Dec 14 2017 18:34:21)

Command line: -if swd -device STM32F767ZI -endian little -speed 1000 -port 2331 -swoport 2332 -telnetport 2333 -vd -ir -localhostonly 1 -singlerun -strict -timeout 0 -nogui
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               on
Init regs on start:            on
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32F767ZI
Target interface:              SWD
Target interface speed:        1000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V9 compiled Jun  2 2222 22:22:22
Hardware: V9.40
S/N: 59401308
Feature(s): GDB, RDI, FlashBP, FlashDL, JFlash, RDDI
Checking target voltage...
Target voltage: 3.28 V
Listening on TCP/IP port 2331
Connecting to target...Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x00000000 (Data = 0xAE37F7DF)
Read 2 bytes @ address 0x00000000 (Data = 0xF7DF)
Received monitor command: speed 1000
Target interface speed set to 1000 kHz
Received monitor command: clrbp
Received monitor command: reset
Resetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x08000328)
Received monitor command: regs
R0 = 00000000, R1 = 00000000, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000, R13= 20080000, MSP= 20080000, PSP= 00000000
R14(LR) = FFFFFFFF, R15(PC) = 08000328
XPSR 01000000, APSR 00000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Received monitor command: speed auto
Select auto target interface speed (2000 kHz)
Received monitor command: flash breakpoints 1
Flash breakpoints enabled
Received monitor command: semihosting enable
Semi-hosting enabled (Handle on BKPT)
Received monitor command: semihosting IOClient 1
Semihosting I/O set to TELNET Client
Received monitor command: SWO DisableTarget 0xFFFFFFFF
SWO disabled successfully.
Received monitor command: SWO EnableTarget 0 0 0x1 0
SWO enabled successfully.
Read 4 bytes @ address 0x08000328 (Data = 0xF7FFB508)
Downloading 1072 bytes @ address 0x08000000 - Verified OK
Downloading 40 bytes @ address 0x08000430 - Verified OK
Downloading 10243 bytes @ address 0x08000460 - Verified OK
Downloading 120 bytes @ address 0x08002C64 - Verified OK
Comparing flash   [....................] Done.
Verifying flash   [....................] Done.
Writing register (PC = 0x080001F8)
Received monitor command: clrbp
Received monitor command: reset
Resetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x08000328)
Received monitor command: regs
R0 = 00000000, R1 = 00000000, R2 = 00000000, R3 = 00000000
R4 = 00000000, R5 = 00000000, R6 = 00000000, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 00000000, R13= 20080000, MSP= 20080000, PSP= 00000000
R14(LR) = FFFFFFFF, R15(PC) = 08000328
XPSR 01000000, APSR 00000000, EPSR 01000000, IPSR 00000000
CFBP 00000000, CONTROL 00, FAULTMASK 00, BASEPRI 00, PRIMASK 00
Reading all registers
Starting target CPU...
Restoring target state and closing J-Link connection...
Shutting down...

```

Cheers
Pei

ilg-ul, 2018-01-20:
> I didn't see any LED on NUCLEO STM32F767ZI blinking...

I'm afraid there is a misunderstanding; all templates create generic projects, and since the wizard did not ask for the board name, it does not know that you want to use that specific NUCLEO board.

Please check the `BlinkLed.h` and update the pin where the LED is connected to match your board.

ilg-ul, 2018-01-25:
fixed since v4.3.2-201801250917

closed 2018-01-25 by ilg-ul