eclipse-embed-cdt/eclipse-plugins #270: System Includes

m4rcs, 2018-01-19:
I'm trying to migrate an old project to the current version of the plugin and that project depends on `-isystem`-includes instead of the normal `-I`. I've have spend several hours looking for this feature but I didn't find it. Am I missing this feature or isn't that supported with that plugin, yet.

Greets, Marc

ilg-ul, 2018-01-20:
I'm not sure what feature you are looking for.

The build plug-in allows to define some of the commonly used compiler options in a graphical way, but does not prevent you to define any other options in the Miscellaneous fields.

How would you like such a feature to be implemented?

m4rcs, 2018-01-22:
I was thinking on a checkbox when defining library path that says "System include".

ilg-ul, 2018-01-22:
I'm afraid such a checkbox is not available in that context.

What can be done, is a new large area, similar to the one used for regular includes.

But then someone else will ask for a similar one for quoted includes... and one for the after system includes... and so on...

Did adding the option in the Miscellaneous field help?

m4rcs, 2018-01-22:
I did not find the Miscellaneous field, yet. Where is it? 

What I did was to edit the "Command line pattern" under C/C++ Build -> Settings -> Tool Settings -> GNU ARM Cross C* Compiler.

ilg-ul, 2018-01-22:
Changing the "Command line pattern" may have unexpected consequences.

Miscellaneous is the last category of options, for each tool (Assembler, Compiler) in the Tool Settings tab.

Click it and in the right side area you'll find an edit named 'Other compiler flags' where you can add anything you need.

ilg-ul, 2018-04-04:
https://gcc.gnu.org/onlinedocs/gcc/Directory-Options.html

added a new section to the Includes categories.

ilg-ul, 2018-04-04:
Fixed on 2018-04-04.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

closed 2018-04-19 by ilg-ul