eclipse-embed-cdt/eclipse-plugins #275: Remove -march=rv32e and add -mabi=ilp32e?

TM1234, 2018-01-25:
At the moment the plugins list -march=rv32e as a configuration option.

However this suggests that to target rv32e one should specify 

`-march=rv32i -mabi=ilp32e`

https://github.com/riscv/riscv-toolchain-conventions/issues/3

Do the plugins need to be changed to remove -march=rv32e and add -mabi=ilp32e?

On the other hand I'm not sure how set in stone the rv32e support, the current implementation in the tools and the way of telling the tools to target rv32e are given that I don't think that the abi is full specified and/or ratified as this suggests?

https://groups.google.com/a/groups.riscv.org/forum/#!topic/sw-dev/nWLvkYTseU4

ilg-ul, 2018-01-25:
I don't know.

The current combination of options in the GUI follows the SiFive requests. 

If you think they are not correct, please discuss with them and decide on a change.

Personally I think rv32e is not a viable specification and should be avoided.

TM1234, 2018-01-26:
Thanks Liviu.
According to Jim here:

https://groups.google.com/a/groups.riscv.org/d/msg/isa-dev/DlRfPVt51Dg/_V-dvfxCBgAJ

the latest on this is that one can use:

`-march=rv32e and -abi=ilp32e`

So in terms of the plugin and the latest tools the arch list is correct but the -mabi=ilp32e option is missing.

But obviously the whole E area is still a bit unclear.
I'll just leave the issue open for further tracking until such time as things settle down a bit.

ilg-ul, 2018-01-27:
> the -mabi=ilp32e option is missing.

so your suggestion is to simply add a new entry to the ABI combo, so we can experiment with the new toolchain additions?

TM1234, 2018-01-27:
Yes - as far as I can tell that's all that's missing from the plugin at the moment.
And, as you say, maybe adding it at least for experimentation might help - even if there seem to be many question marks over the direction and current implementation of rv32e support at the moment... :-|

ilg-ul, 2018-01-30:
please try the pre-release version 4.3.3-201801301548 from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-experimental

TM1234, 2018-01-30:
Thanks Liviu - yes, I see the -mabi=ilp32e option in this version.
I'm not sure which, if any, version of the tools supports this though.
Certainly not the last release: https://github.com/riscv/riscv-gnu-toolchain/releases/tag/v20171231
With that I get:

```
riscv64-unknown-elf-gcc: error: unrecognized argument in option '-mabi=ilp32e'
riscv64-unknown-elf-gcc: note: valid arguments to '-mabi=' are: ilp32 ilp32d ilp32f lp64 lp64d lp64f; did you mean 'ilp32'?
```

But I guess that's not your problem once the plugin lists the option?

ilg-ul, 2018-02-01:
I don't know what to say, I would delay adding this option until really available.

TM1234, 2018-02-01:
I would be inclined to leave it there in case somebody builds a bleeding edge toolchain (from the master branch?) that includes the latest rv32e support and wants to play with it using the plugins?.

ilg-ul, 2018-02-01:
Ok, I'll leave it there, but I have to explicitly mention in the release note that this option is not yet supported by the production toolchains. even so, I might get complaints...

as for the brave ones who build their own bleeding edge toolchain, I doubt they ever touch Eclipse...

ilg-ul, 2018-04-03:
Fixed on 2018-01-30.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

closed 2018-04-19 by ilg-ul