eclipse-embed-cdt/eclipse-plugins #276: include some other folders?

jiapei100, 2018-01-26:
@ilg-ul 

Hi, I actually created a folder named **led**, in which I put **ONLY** two files, named **led.h** and **led.c** respectively. And I'm pretty sure I modified **main.c** accordingly. Afterwards, I **Build Project**, almost successful, but with an **ERROR** message 
`undefined reference to `LED_GPIO_Config'`

However, if I put **led.h** under the project **include** folder, and **led.c** under the project **src** folder, I **Build Project** successfully, with **NO** **ERROR** message.

I wonder if I use this **GNU-MCU-Eclipse** plugin, will the project look on any created folder as a single library? I  mean, for instance, in my case above, will the Eclipse project build the folder **led** with two files **led.h** and **led.c** into a library for the **main.c** to link finally? Otherwise, why there is an **ERROR** message **undefined reference** ?

Cheers
Pei

ilg-ul, 2018-01-26:
did you tell Eclipse that the folder `led` contains source files?

jiapei100, 2018-01-26:
@ilg-ul 
I did put two files **BlinkLed.h** and **BlinkLed.c** under the folder **BlinkLed** physically. I mean, I just did this **copy/move** using command line, and **refresh**ed it within **Eclipse IDE**.

How can I **tell** Eclipse that the folder contains source files? I just put one **.h** and one **.c** files under that particularly folder....

ilg-ul, 2018-01-26:
Eclipse processes only files in folders defined as Source folders and ignores all other folders.

As homework, you have to discover how to define the Source folders.

closed 2018-01-26 by ilg-ul