eclipse-embed-cdt/eclipse-plugins #277: undefined reference to `HAL_DMA_Init'

jiapei100, 2018-01-27:
Hi, @ilg-ul 

Now, every single file is successfully compiled, but the project failed to build, with a lot of **undefined reference to** **ERROR** messages. In fact, some file calls, for example, **HAL_DMA_Init()** function, which is declared/defined in file **system/include/stm32f7-hal/stm32f7xx_hal.dma.h**.

I wonder if I need to link some **stm32f7xx_hal** libraries? in order to successfully have the project built?

And, where(which directory) can I find those libraries?

Cheers
Pei

ilg-ul, 2018-01-27:
for support please use the project forum.

closed 2018-01-27 by ilg-ul

jiapei100, 2018-01-27:
@diabolo38 
Hey, do you mean the software downloaded from [https://my.st.com/content/my_st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-mcu-packages/stm32cubef7.html](https://my.st.com/content/my_st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-mcu-packages/stm32cubef7.html) ?

Yup, I had it installed, but it seems there are **ONLY** **.a** static libraries, but **NO** **.so** dynamic libraries?

Cheers
Pei

jiapei100, 2018-01-27:
@ilg-ul 
Thank you @ilg-ul . I posted a **NEW** question at [https://www.element14.com/community/message/234249/l/how-can-i-add-those-grayed-system-hal-files-into-my-stm32f7-project-for-build#234249](https://www.element14.com/community/message/234249/l/how-can-i-add-those-grayed-system-hal-files-into-my-stm32f7-project-for-build#234249) .