eclipse-embed-cdt/eclipse-plugins #278: stm32f407 Peripherals view

t-szczyrba, 2018-01-27:
### Description

Peripherals view is empty as described in https://www.element14.com/community/thread/62731/l/peripherals-view-empty-with-stm32f407

### Steps to Reproduce

Choose device STM32F407VG with pack 2.11.0 from STM32F4xx_DFP from keil.com, enter debug perspective, open Peripherals view 

**Expected behaviour:** [What you expected to happen]

In the peripherals view there should be a list of peripheral registers of STM32F407VG

**Actual behaviour:** [What actually happened]

No registers visible  in the list - list empty

### Versions

* [plug-in version]
4.3.2.20180125

* [Eclipse version]
Oxygen.2 Release (4.7.2)

* [Java version]
1.8.0_162

* [operating system]
opensuse 42.3

* [toolchain version]
gcc-arm-none-eabi-7-2017-q4-major also tested with gcc-arm-none-eabi-6-2017-q1-update

ilg-ul, 2018-01-27:
For completeness of your description, you also need to assign the STM32F407VG to the project.

bureaua, 2018-01-28:
Same issue on stm32f1
After Eclipse and plugin installation from scratch on:
Win10_x64 -- works fine, but after some time Peripherals view is empty
Ubuntu 16.04 x64 -- Peripherals view is empty

ilg-ul, 2018-01-28:
yes, thank you for the bug report, I'll try to fix it these days.

bureaua, 2018-01-28:
msg which I see in terminal (Ubuntu):
```
Warning: NLS unused message: TabSvd_group_text in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
Warning: NLS unused message: TabSvd_label_text in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
Warning: NLS unused message: TabSvd_label_tooltip in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
Warning: NLS unused message: TabSvd_button_Browse_text in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
Warning: NLS unused message: TabSvd_button_Variables_text in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
Warning: NLS missing message: PeripheralsView_FileAction_text in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
Warning: NLS missing message: PeripheralsView_FileAction_toolTipText in: ilg.gnumcueclipse.debug.gdbjtag.ui.messages
```

ilg-ul, 2018-01-30:
I fixed this, please install the pre-release from

 http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test

bureaua, 2018-01-31:
Many Thanks!
Looks like everything is working fine

ilg-ul, 2018-01-31:
great!

good luck with your projects!

t-szczyrba, 2018-01-31:
I'll check it when I am back in the office. Many thanks

MatteoMilandri, 2018-03-24:
Can confirm that this has been happening on the nRF52840 as well. Thanks for the fix! I'll wait for the 4.3.3 release.

PheeL79, 2018-04-03:
The same issue.
No actions after the message:

Parsing SVD file "/home/xxx/Packages/NordicSemiconductor/nRF_DeviceFamilyPack/8.15.2/SVD/nrf52.svd"...

No message about successful parse "on xx ms" appear.
"Peripherals" view is empty on debugging session.
Regular core registers are visible through "Registers" view.
Suggest this is a *.svd plug-in parser issue or not supported SVD file format from Nordic.

[plug-in version]
4.3.1.201801301548

[Eclipse version]
Neon.3 Release (4.6.3)
Build id: 20170314-1500

[Java version]
java version "1.8.0_161"
Java(TM) SE Runtime Environment (build 1.8.0_161-b12)
Java HotSpot(TM) 64-Bit Server VM (build 25.161-b12, mixed mode)

[operating system]
ubuntu 4.13.0-37-generic x86_64

[toolchain version]
gcc-arm-none-eabi-6-2017-q2-update

ilg-ul, 2018-04-03:
@PheeL79, did you install the version from updates-test?

PheeL79, 2018-04-03:
@ilg-ul As you can see, version of mine is newer 4.3.1.201801301548 (actually, while I tried all possible solutions, I've installed from the updates-experimental). And because of that Eclipse refuse me to downgrade to the lower version. I don't know may I force it to do such an action, but it didn't work on the 1501 revision either while I tried.

The message from Eclipse:
Cannot complete the request.  See the error log for details.
"GNU MCU C/C++ Generic Cortex-M Project Template" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ Packs (Experimental)" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ STM32Fx Project Templates" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ PyOCD Debugging" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ OpenOCD Debugging" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ J-Link Debugging" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ Documentation (Placeholder)" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ CodeRed Debug Perspective" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ ARM Cross Compiler" will be ignored because a newer version is already installed. 
"GNU MCU C/C++ QEMU Debugging" will be ignored because a newer version is already installed. 

Can you, please, manually test the plug-in's parser on the exact nrf52.svd file? I strongly suspecting it.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

closed 2018-04-19 by ilg-ul