eclipse-embed-cdt/eclipse-plugins #279: Not refreshing layout when switching between ARM configuration and RISC-V configuration inside single project.

AntonKrug, 2018-02-05:
### Description

A project which has to target 2 platforms shows strange behaviour when switching configurations.

### Steps to Reproduce

1. Import to RISCV project a configuration from ARM project so you could be targeting both platforms by quickly switching.
2. In project settings now the RISCV is showing as RISCV.
3. By clicking on the top to switch configuration, now I can switch to ARM and the layout updates,
4. Switching back to RISCV project will still keep the ARM layout and not update to RISCV.

**Expected behaviour:** 

Switching between configuration should always update to the corresponding target,

**Actual behaviour:**

[image]
[image]
[image]

### Versions

* [plug-in version] 2.6.1
* [Eclipse version] 4.7.2
* [Java version] 9.0.4
* [operating system] Windows 10 64 bit
* [toolchain version] riscv 7.2.0, arm 7.2.1

ilg-ul, 2018-04-05:
@AntonKrug, I'm not able to reproduce this problem.

I tried with the latest version of the plug-ins and also with an older version.

I creted a new RISC-V project and imported the Debug configuration from an ARM project, saved the new configuration and then entered again in the Properties windows. Switching from ARM to RISC-V and back works as expected.

I think the step to save everything after adding the new configuration is mandatory, otherwise Eclipse might not be aware of the new configuration.

ilg-ul, 2018-04-06:
If you can reproduce this, please come back with more details. Otherwise please close this issue.

ilg-ul, 2018-04-06:
Close the Properties page and enter again.

AntonKrug, 2018-04-06:
Ah, yes been doing that and even closing and opening eclipse doesn't help. Importing ARM to RISCV and vice versa is not fixing it as well.

ilg-ul, 2018-04-06:
Does it happen on Linux too?

AntonKrug, 2018-04-06:
No Linux works fine even without the "save".

ilg-ul, 2018-04-06:
Then I doubt it is a problem with the plug-in.

AntonKrug, 2018-04-06:
Can you replicate it on Windows?

ilg-ul, 2018-04-06:
Yes. But I have no idea how to address this. As you know, the Tool Settings tab is managed by CDT, based on the toolchain definitions in plugin.xml; I have no control on the redraw logic there.

ilg-ul, 2018-04-06:
If you can install other regular (non cross) toolchains on your windows, do the same test, and if the behaviour is the same, report it to CDT.