eclipse-embed-cdt/eclipse-plugins #281: Tune drop-down list doesn't support "size" and does contain invalid "generic" option.

AntonKrug, 2018-02-23:
### Description

[image]

Making RISCV project and using the tune option "generic" with new riscv gcc will cause this error on compile time:
"cc1.exe: error: unknown cpu 'generic' for -mtune"
On another hand there is the "size" tune option which is no available from the menu:
https://github.com/riscv/riscv-gcc/blob/riscv-gcc-7.2.0/gcc/config/riscv/riscv.c#L339

If the -mtune generic worked in the past then plugin should probably detect the toolchain version and give it valid options to select depending on the toolchain version. 

Or give the user all possible options from all version "rocket/generic/size" and leave it to the user to know which one is usable. But with new gcc toolchain, in essence, there is no option now, only to use rocket, which allows misaligned memory access. This option expects the trap handler to handle that memory access, if that handler is not meant to handle such requests then the code will crash un runtime.

The blog makes it sound less harmful than it really can be:

https://www.sifive.com/blog/2017/08/14/all-aboard-part-1-compiler-args/

"The last compiler argument that’s involved in specifying a target is the simplest of the bunch. While the -march argument can cause systems to be unable to execute code and the -mabi argument can cause objects to be incompatible with each other, the -mtune argument should only change the performance of the generated code. As it currently stands we really don’t have any tuning models for RISC-V systems. Unless you’ve just added a new tuning parameter to our GCC port, you probably shouldn’t bother doing anything with this argument."

### Versions

* [plug-in version] 2.6.1
* [Eclipse version] 4.7.2
* [Java version] 9.0.4
* [operating system] Windows 10 64 bit
* [toolchain version] riscv 7.2.0, arm 7.2.1

ilg-ul, 2018-02-23:
do you suggest to remove 'generic' and to add 'size'?

AntonKrug, 2018-02-23:
Yes, or if you want to keep 'generic' at least add 'size'.

ilg-ul, 2018-02-23:
ok, I'll consider it for the next release.

AntonKrug, 2018-02-23:
OK, thank you. Would be a separate checkbox for 'mstrict-align' possible as well? For somebody who wants to keep rocket optimizations but has much simpler trap handler, which is very likely to happen.

Users might be affected without realizing it and problems occurring only when some optimizations enabled. Or the fact that only some structures might trigger assembler to do unaligned access so the problem might not be discovered straight away. Which I think makes it even more dangerous and hard to spot.

ilg-ul, 2018-02-23:
makes sense.

ilg-ul, 2018-04-03:
Added -mtune=size.
Marked -mtune=generic as deprecated.
Added -mstrict-align & -mno-strict-align as combo.

ilg-ul, 2018-04-03:
Fixed on 2018-04-03.

AntonKrug, 2018-04-03:
> Fixed on 2018-04-03.

Is there a branch/commit you want me to test/verify?

ilg-ul, 2018-04-03:
I'll let you know when the pre-release will be available on updates-test.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

AntonKrug, 2018-04-06:
Tested on Linux/Windows, it works as advertised. Adding the align options was a good idea. Thank you very much.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

closed 2018-04-19 by ilg-ul