eclipse-embed-cdt/eclipse-plugins #284: [Request] Tool Settings for Unmanaged Makefile projects

tcwan, 2018-03-26:
### Description

The closest issue I found was [this](https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/5). I'd like to configure the Tool Settings for specifying the Target Processor in imported Makefile projects (In the previous issue they were referred to as Unmanaged Projects). 

### Steps to Reproduce

1. File->Import->C/C++->Existing Code as Makefile Project
2. Configure Project 
3. C/C++ Build->Settings->Tool Settings tab not available.

**Expected behaviour:** [What you expected to happen]

To be able to configure Tool Settings Target Processor even for unmanaged projects.

**Actual behaviour:** [What actually happened]

Tool Settings can only be enabled if I enable "Generate Makefiles automatically" in C/C++ Build dialog "Builder Settings" tab. Enabling this feature will cause [my project](https://github.com/tcwan/ev3dev-arm-bbr) build to fail since the project contains subprojects which should not be rebuilt except under specific conditions.

### Versions

* GNU MCU Eclipse 4.3.2
* Eclipse CDT Oxygen.2
* JDK 9
* macOS High Sierra
* Docker-based Debian Stretch cross-toolchain `arm-linux-gnueabi-*` (requires Eclipse Docker Tooling and [Docker image](https://hub.docker.com/r/ev3dev/debian-stretch-cross/)

tcwan, 2018-03-26:
FYI, I've documented some of the steps needed to setup Eclipse for cross-compilation with Docker [here](https://github.com/tcwan/edu-docs/blob/master/docs/programming/eclipse-ide.md).

tcwan, 2018-03-26:
Hmm. How are the settings defined? e.g. C_FLAGS, CPP_FLAGS, etc.
I can reference them from the Makefile if I know what they are and they'll be used to override the defaults.

ilg-ul, 2018-03-26:
The settings are internal to Eclipse, they are not reflected out in the environment, so you cannot reference them in the Makefile.

tcwan, 2018-03-26:
Incidentally, in Imported projects, the "Toolchains" tab doesn't list the other GNU tools such as ar, make, size, etc. unlike for case of Managed projects. Only C compiler (gcc) and C++ Compiler (g++) can be specified. Not sure if this is an omission or not.

tcwan, 2018-03-26:
> The settings are internal to Eclipse, they are not reflected out in the environment, so you cannot reference them in the Makefile.

I see. Is there another way to solve my problem using Managed builds? I don't know how much leeway I have in modifying Makefiles in Eclipse for managed builds. Would Eclipse overwrite the customized makefiles?

ilg-ul, 2018-03-26:
The Eclipse managed build process recreates the build trees in separate folders like Debug/Release.

It parses all source folders for files with known extensions and builds them all. The list of source folders is under your control. You can exclude from build any files in the sources folder.

The GNU MCU Eclipse build plug-ins can handle only managed projects.

jantje, 2018-03-26:
@tcwan 
The naming is confusing and what you are referring to is out of @ilg-ul control.
Let me explain how I see the situation at hand. 

There are 2 builders (stuff that convert code in executables) in CDT

1. Managed build->CDT manages the build for you
2. Unmanaged builds->Someone maintains a makefile to build (probably you)

Next to these builders runs the **Indexer**. The indexer takes care of plenty of the gui related stuff and in my eyes makes CDT. 

Even though the indexer is so vital for CDT most people do not realize it's existence (read CDT did a great integration job). Once you bump into "proof of the indexer existence" it is often quite hard to "get it" 

The "proof of existence" you bumped into is the "indexer configuration". Let me explain:

To work properly the indexer needs to know compiler defined defines and include paths. This info is defined in the compiler and not in ar, make, size.
In managed build the indexer asks the location and name of the compiler to the managed build engine.
In unmanaged build the indexer asks the user. As only the compiler is needed (not  ar, make, size) only the  compiler is requested.

ilg-ul, 2018-03-26:
For the indexer to be accurate, it needs to know the exact command line used to compile each file, which plain make builds cannot provide.

jantje, 2018-03-26:
> For the indexer to be accurate, it needs to know the exact command line used to compile each file, which plain make builds cannot provide.

Indeed: the old adagio still counts: "garbage in is garbage out"

tcwan, 2018-03-28:
Ok. The explanations help me understand the situation better now.
I guess it is a 'all-or-nothing' situation with respect to controlling the build parameters.
I'll close this issue.

closed 2018-03-28 by tcwan