eclipse-embed-cdt/eclipse-plugins #285: Illegal PDSC file size -1 during Packs update

bureaua, 2018-03-31:
### Description
A lot of messages with "Illegal PDSC file size" during Packs update
[image]

### Steps to Reproduce
- Download and run eclipse-oxygen-3-x86_64
- Open C/C++ Packs (Show View)
- Press update button on Packs tab

## Versions
[plug-in version]
4.3.1.201801301501

### [Eclipse version]
Oxygen.3 Release (4.7.2)

### [Java version]
java 9.0.4

### [operating system]
Ubuntu 16.04 x64
Windows 10 x64

ilg-ul, 2018-03-31:
yes, unfortunately sometimes the servers do not respond and the plugin reports errors.

bureaua, 2018-03-31:
Will it possible to ignore this message in automatic way? Because I have to press now "Ignore" button each time, many times to continue.

ilg-ul, 2018-03-31:
If the plugin will automatically ignore them, you'll think that everything went well, and you'll be surprised when those packs have no content, or were not updated.

just curious, during the lengthy update process, did you check the messages in the Packs Console?

bureaua, 2018-03-31:
> in the Packs Console?
```
Missing "/opt/eclipse-oxygen-3-x86_64/Packages/.cache/NXP.QN908x_DFP.1.1.4.pdsc", ignored by user request.
Processing "/opt/eclipse-oxygen-3-x86_64/Packages/.cache/Keil.MCB1500_BSP.1.2.0.pdsc" for content.xml ...
Schema version "1.2".
```
Should I expect something interesting?

> you'll think that everything went well

Agree. May be Check Button for this, like "ignore this message in future"...

ilg-ul, 2018-04-03:
I added an 'Ignore All' button.

ilg-ul, 2018-04-03:
Fixed on 2018-04-03.

bureaua, 2018-04-04:
By using which link I can try to test ?
http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test -- does not see any updates

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

closed 2018-04-19 by ilg-ul