eclipse-embed-cdt/eclipse-plugins #286: SVD file type identification fails

ilg-ul, 2018-04-03:
In certain conditions, SVD files are not recognised, for example nrf52.svd for nRF52832_xxAA.

(the problem was reported by @PheeL79 at the end of https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/278).

ilg-ul, 2018-04-03:
Fixed on 2018-04-03.

PheeL79, 2018-04-03:
As a temporal workaround, you can convert the *.svd file and save it with "UTF-8" option in various text editors. After that the issue is gone (all the peripheral registers are visible and modifiable).
Thank you for the support!

ilg-ul, 2018-04-03:
Yes, this can be a functional workaround.

Anyway, the v4.3.3 is planned for the next weekend.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

closed 2018-04-19 by ilg-ul