eclipse-embed-cdt/eclipse-plugins #290: Hide the Devices tab for configurations not managed by GME

ilg-ul, 2018-04-05:
The Devices tab contributed by the managedbuild.packs plug-in should not be visible for build configurations not managed by GME.

ilg-ul, 2018-04-05:
To identify if the configuration is managed by GME, add a hidden option and test it (by SuperId("ilg.gnumcueclipse.managedbuild.cross.option.base.showDevicesTab")).

ilg-ul, 2018-04-05:
Fixed on 2018-04-05.

ilg-ul, 2018-04-05:
The 4.3.3.201804051758 pre-release is now available for testing from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates-test.

ilg-ul, 2018-04-19:
Fixed since 4.3.3.201804191501.

ilg-ul, 2018-05-31:
Reopened due to #303.

ilg-ul, 2018-05-31:
The requirement for a managed project was relaxed since 4.3.4, 2018-05-31.

ilg-ul, 2018-05-31:
@nolange, please check the version from updates-test and let me know if the Devices tab is now visible in your setup.

nolange, 2018-06-01:
Nope, still need to pick "ARM Cross GCC" as compiler, test-setup as specified in #295

ilg-ul, 2018-06-01:
Yes, you need to use one of the known cross toolchains (ARM or RISC-V), but you no longer need a managed project.

nolange, 2018-06-01:
I don't know what constitutes a "managed project", it worked for me before (using cmake4eclipse Builder). The only difference is that I don't get errors if the ARM/RISC-V Toolchains are not installed.
Whats new however is, that entering the toolchain settings page locks up eclipse for ~25 seconds.

ilg-ul, 2018-06-01:
A "managed project" is a project that Eclipse knows in such detail that it is able to build automatically, either directly, using the now deprecated internal builder, or by generating the makefiles and invoking make.

cmorty, 2018-06-12:
Sorry for the bugspam, but I wasn't able to find this out on my own: When will 4.3.4 be released and are there nightly builds?

TM1234, 2018-06-12:
> are there nightly builds?

I don't know about nightly builds but see here about pre-release and experimental versions:

https://gnu-mcu-eclipse.github.io/plugins/download/#pre-release-versions

Hope this helps.

ilg-ul, 2018-06-12:
4.3.4 will be released with the next Eclipse release, probably in July.

For now you can use it from the updates-test URL, as Tommy mentioned.

cmorty, 2018-06-12:
OT: Thanks a lot. See #306  :)

closed 2018-07-21 by ilg-ul